emelianov/modbus-esp8266 #162: Problem with pushCoil

hagbrain21, 2021-11-04:
Hello and thank you for this library!
Everything works really nice an reliable, except one thing.
I build a TCP/RTU bridge and pullIsts are working from RTU to TCP.
My problem is the pushCoil command that isnt working. I checked the existing issues and checked everything double in my code and hardware.
Here is a part of my code. I only use one of the lines under case 1:

if (!mbrtu.slave()) {
    switch(cReg++) {
    case 0:
            mbrtu.pullIsts(1, 200, 200,8); //WORKING
    break;
    case 1: 
            //mbrtu.pushCoil(2, 500, 500); //NOT WORKING
            //mbrtu.writeCoil(2, 500, mbip.Coil(500)); //WORKING
    break;
    default:
       cReg = 0;
    }
  }

Where is my problem?
Thanks so far.

emelianov, 2021-11-04:
Hello,
It's my fault. Fixed in e48d76c.

Probably also you would interested in testing of new bridging support:
https://github.com/emelianov/modbus-esp8266/blob/truebridge/examples/Bridge/true/true.ino

hagbrain21, 2021-11-04:
Thank you for the fix. Ill check the new library. :-)

closed 2021-11-04 by hagbrain21