emelianov/modbus-esp8266 #163: Check value incoming from master - help wanted

Naguco, 2021-11-05:
Hi Emiliano!
First of all, thank you for your work in this library and congratulations!
I want to ask you a question.
Right now I am programming a slave device with the esp8266 and your modbus library. 
I have a set of registers (from 0 to 19) wich can be read by modbus protocol.
The point is that when the master sends me (slave) a 0x06, I have to check the value that it is sending to me to, if necessary, return an EX_ILLEGAL_VALUE  code.
I saw that I can add callbacks to check the addresses (already implemented in my code) and return ILLEGAL_ADDRESS if necessary, but i don't find an example to return an ILLEGAL_VALUE code.

Sorry if this question is too basic, but I am facing with this issue since yesterday and I don't really know how to handle this situation...

Thank you in advance!

emelianov, 2021-11-05:
Hello,
The only way to achieve required result is [orRequest callback](https://github.com/emelianov/modbus-esp8266/blob/master/examples/Callback/Request/Request.ino). For write single reg it's very suitable.

```c
typedef Modbus::ResultCode (*cbRequest)(Modbus::FunctionCode fc, const Modbus::RequestData data);
bool onRequest(cbRequest cb = _onRequestDefault);
bool onRequestSuccess(cbRequest cb = _onRequestDefault);

union Modbus::RequestData {
    struct {
        TAddress reg;
        uint16_t regCount;
    };
    struct {
        TAddress regRead;
        uint16_t regReadCount;
        TAddress regWrite;
        uint16_t regWriteCount;
    };
    struct {
        TAddress regMask;
        uint16_t andMask;
        uint16_t orMask;
    };
};
```

Callback function receives Modbus function code, structure `Modbus::RequestData` containing register type and offset (`TAddress` structure) and count of registers requested. The function should return [result code](#Result codes *Modbus::ResultCode*) `Modbus::EX_SUCCESS` to allow request processing or Modbus error code to block processing. This code will be returned to client/master.

Another way is to return original register value in onSet callback. But error code returned is `Modbus::EX_SLAVE_FAILURE` that case. For write multiple registers I see no other easy way. In addition `Modbus.cpp` may be patched to return desired error code (and it's good point to make it configurable)

Naguco, 2021-11-05:
Thank you for your time!

Right now what I have implemented is this callback function for the onRequestData, which depending on the number of the register, I send to the master an  ILLEGAL_ADDRESS or a SUCCESS, but I also need a check for the data that the master sends to me.. IE: I only accept for the register 1 a value of 0 or 1, if you send me any other value, I have to send you ILLEGAL_VALUE. There is no way to get the value on this first callback "onRequest(cbRequest cb = _onRequestDefault);"

Thank you again!

Naguco, 2021-11-05:
Ok, I think that I found the way! Thank you so much for the info!

closed 2021-11-05 by Naguco