emelianov/modbus-esp8266 #164: Temos D1 Mini

thalesmaoa, 2021-11-18:
I've been trying different types of libraries and yours seems to be the best one.
Unfortunately, I can't get it to work properly with Wemos D1 Mini.

It always return 0xE2.
I'm not sure why this is happening. I'm out of clue how to test it.
My RS485 line is ok (works with Arduino Mega).
My SoftwareSerial works (tested with PC).

I'm lost...

emelianov, 2021-11-19:
```c
EX_ILLEGAL_ADDRESS      = 0x02, // Output Address not exists
```
For the first it means that wiring is okay. It's probably because address base (0 vs 1). The library uses 0.
Try to subtract 1 from currently used register address. Also if target address is written like 40001 try to use 1 (or 0) as register address.

thalesmaoa, 2021-11-21:
Thanks! After your post I did a lot of tests. Unfortunetly it seems to be a hardware problem. Not sure why, but when I use RS485, Wemos D1 mini with a level shifter, It does not communicate with the device.

emelianov, 2021-11-22:
Sorry I've misread your first message. You are right E2 errors caused by hardware errors.
Indeed all the RS-485 transceivers I've tested was working well with 3.3V logic directly without any level shifter.

closed 2021-11-22 by thalesmaoa