emelianov/modbus-esp8266 #165: FeatureRequest: Allow User parameters for callbacks

kscholty, 2021-11-24:
In order to allow re-using callback functions without the need for global status variables it would be great if it was possible to attach a user defined value (usually void*) to a read request. This value should then be passed to the callback when called.

How to do that: 
1) One new parameter of type void * to the readXXX functions,
2) Add parameter of type void* to master callbacks
3) Store the user value and pass it on in  ModbusRTUTemplate::task() and ModbusTCPTemplate()

emelianov, 2021-11-27:
How about std::bind ? Something like
```c
#include <ModbusRTU.h>

#define SLAVE_ID 1
#define FIRST_REG 0
#define REG_COUNT 2

ModbusRTU mb;

bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data, uint16_t t) { // Callback to monitor errors
  if (event != Modbus::EX_SUCCESS) {
    Serial.print("Request result: 0x");
    Serial.print(event, HEX);
  }
  return true;
}

uint16_t parm = 0;
using namespace std::placeholders;
static auto f = std::bind(cb, _1, _2, _3, parm);

void setup() {
  Serial.begin(115200);
  Serial1.begin(9600, SERIAL_8N1, 25, 32);
  mb.begin(&Serial1);
  mb.master();
}

uint16_t res[REG_COUNT];

void loop() {
  if (!mb.slave()) {    // Check if no transaction in progress
    mb.readHreg(SLAVE_ID, FIRST_REG, res, REG_COUNT, f); // Send Read Hreg from Modbus Server
    while(mb.slave()) { // Check if transaction is active
      mb.task();
      delay(10);
    }
    Serial.println(res[0]);
    Serial.println(res[1]);
  }
  delay(1000);
}
```

closed 2021-12-20 by emelianov

dautushenka, 2023-10-17:
> How about std::bind ? Something like
>
> ```c
> #include <ModbusRTU.h>
>
> #define SLAVE_ID 1
> #define FIRST_REG 0
> #define REG_COUNT 2
>
> ModbusRTU mb;
>
> bool cb(Modbus::ResultCode event, uint16_t [...]
The `std::bind` does not work on arduino devices (

emelianov, 2023-10-19:
> unfortunately, the `std::bind` does not work with arduino devices ( @emelianov FYI

It's device-depended indeed.