emelianov/modbus-esp8266 #167: Compiling error

zinkett, 2021-12-14:
Hi, 
I'm trying your library with ESP32 + W5500 with your example ([examples/TCP-Ethernet/server/server.ino](https://github.com/emelianov/modbus-esp8266/blob/master/examples/TCP-Ethernet/server/server.ino))
It compiles and work correctly but if I add `#include <WiFi.h>` (for add wifi function..) I've a complilation error: 

`.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwip/inet.h:142:58: error: expected identifier before '(' token
#define inet_aton(cp, addr)             ip4addr_aton(cp, (ip4_addr_t*)addr)`
`.pio\libdeps\esp32dev\Ethernet\src/Dns.h:21:43: error: expected ',' or '...' before 'IPAddress'
  int inet_aton(const char *aIPAddrString, IPAddress& aResult);`

can you please help me?
Thanks.

emelianov, 2021-12-20:
Hi,

It looks like incompatibility between Ethernet library and ESP32 LwIP stack.
As a workaround replace Ethernet.h with 
```c
#pragma once
#include "ModbusAPI.h"
#include "ModbusTCPTemplate.h"

// Ethernet class wrapper to be able to compile for ESP32
class EthernetServerWrapper : public EthernetServer {
    public:
    EthernetServerWrapper(uint16_t port) : EthernetServer(port) {

    }
    void begin(uint16_t port=0) {

    }
};

class ModbusEthernet : public ModbusAPI<ModbusTCPTemplate<EthernetServerWrapper, EthernetClient>> {};
```

emelianov, 2022-03-08:
Fixed in dfb1ce8

closed 2022-03-08 by emelianov