emelianov/modbus-esp8266 #168: Use of vTaskDelay(0); ?

gonzabrusco, 2021-12-19:
Hello!

This is not an issue, it's a question about the logic behind the code. 

Could you explain me why you are using vTaskDelay(0); in ModbusRTU.cpp in replacement for taskENTER_CRITICAL and vTaskSuspendAll() ?? 

How does this work?

Thanks
Gonzalo

emelianov, 2021-12-20:
Hello  Gonzalo,

taskENTER_CRITICAL was removed as it's hangs ESP32S2 (C3 have't tested but I'm afraid it's also affected). Also no way to determinate ESP32/S2/C3 is the build target at compile time.
 
The idea of using vTaskDelay(0) is following:
RTOS by default allows task to run 100msec before task to be suspended to give a time for other tasks.
vTaskDelay(0) forces task to suspend and switch to other tasks execution.
After exit from vTaskDelay(0) we have whole 100msec time slot to put serial data to buffer, flush buffer and operate direction control pin (if needed) without interruption.
For sure we still limited to ~90 bytes frame size for 9600 but it's best we can do without deep dive to RTOS programming.

gonzabrusco, 2021-12-20:
Clear enough. Thanks for the explanation!

closed 2021-12-20 by gonzabrusco