emelianov/modbus-esp8266 #173: "Request result: 0xE4, Mem: 375624" [ESP32][RS-485 transreceiver]

puneethintellithink, 2021-12-28:
Hello guys,

Been trying to debug this since ages. Please help me. I understand it is a timeout error but I'm not able to figure why the library is throwing timeout error at me. Wiring, Code, everything attached below. Please let me know if you need more details. I will share them ASAP.

Trying to read registers for an energy meter [MFM384](http://www.selectautomation.net/downloads/mfm384_operating.pdf) using RS-485 and [TD321D485H](https://www.mornsun-power.com/html/pdf/TD321D485H-A.html) transreceiver.

I'm trying to read the following registers:
40007 - **SLAVE ID**
Type- **Integer**
Size - **1**

Settings on the energy meter:
Network - **Single Phase (V1)**
BaudRate - **9600**
Parity - **Even**
StopBits - **1**

Using the code below:
`
/*Includes*/
#include <ModbusRTU.h>

/*Definitions*/
#define BAUD_RATE 9600  
#define RS485_MODE 15
#define SLAVE_ID 1

/*Instances*/
ModbusRTU mb;
//SoftwareSerial mySerial(RS485_RX, RS485_TX);
/*Instances END*/

/*Variables START*/
uint16_t myWord;
/*Variables END*/

//Modbus Call back function for ERROR
bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) 
{ // Callback to monitor errors
  Serial.printf_P("Request result: 0x%02X, Mem: %d\n", event, ESP.getFreeHeap());
  return true;
}

/*Void Setup START*/
void setup() 
{
  Serial.begin(BAUD_RATE);
  Serial2.begin(BAUD_RATE, SERIAL_8E1);
  
  Serial.println();

  Serial.println("START");
  Serial.println();

  mb.begin(&Serial2, RS485_MODE);
  mb.master();
}
/*Void Setup END*/

/*Void Loop START*/
void loop() 
{ 
  mb.readHreg(SLAVE_ID , 7, &myWord, 1, cb);//Address 40007 for displaying SLAVE ID (check the energy meter datasheet above)
  Serial.print("DATA: ");
  Serial.println(myWord3);
  mb.task();
  //yield();
  delay(500);
}
/*Void Loop END*/`

Block Diagram:
[image]

Transreceiver connections:
B - RS-485 -ve
A - RS-485 +ve
TX - Serial2 RX
RX - Serial2 TX

puneethintellithink, 2021-12-28:
@emelianov Request you to share your 2 cents on this

ArminPP, 2021-12-30:
Hi,
I am not a ModBus specialist,
but the manual of your sensor on pages 26 and 47 says :
> Settings on the energy meter:
> Network - Single Phase (V1)
> BaudRate - 9600
> Parity - None                  <-----------------
> StopBits - 1

So I suggest, you should set 
`Serial2.begin(BAUD_RATE, SERIAL_8E1);` to `Serial2.begin(BAUD_RATE, SERIAL_8N1);`

.
I am not familiar with the second paramerter of
`mb.begin(&Serial2, RS485_MODE);`
but as I understand the library, you hand over that the txPin is N°15 (RS485_MODE) ...?
In my little test programs I only use 
```
  mb.begin(&Serial2);   
  mb.master();
```
and it still works.

After a look at the datasheet of the TD321D485H transreceiver I do not see a pin to connect with 'RS485_MODE' ?
On page 3 there is a pretty straight forward schematic, where you can only connect rx/tx gnd & vcc ...
[image]

puneethintellithink, 2021-12-30:
> Hi, I am not a ModBus specialist, but the manual of your sensor on pages 26 and 47 says :
>
> So I suggest, you should set `Serial2.begin(BAUD_RATE, SERIAL_8E1);` to `Serial2.begin(BAUD_RATE, [...]

I setup the energy meter with new parameters as mentioned below and written in my code, viz,
Network - Single Phase (V1)
BaudRate - 9600
Parity - Even
StopBits - 1

> . I am not familiar with the second paramerter of `mb.begin(&Serial2, RS485_MODE);` but as I understand the library, you hand over that the txPin is N°15 (RS485_MODE) ...? In my little test programs [...]

I might have attached a wrong PDF. It's updated now. Please re-click on the link to find the RS-485 module's application notes. As you go through the pdf, you will find the following typical application circuit on page 3.
[image]

ArminPP, 2021-12-31:
OK, I see...

Unfortunately I can't help with this issue.

First I thought, that 8N1 is mandatory with Modbus, but the specs allow also 'even' or 'odd' parity checking. (https://modbus.org/docs/Modbus_over_serial_line_V1_02.pdf)

I also had (have) some troubles with my application, to dig in deeper I bought some cheap USB to RS485 converter.
[image]
(eg. https://www.banggood.com/de/USB-To-RS485-Converter-USB-485-With-TVS-Transient-Protection-Function-With-Signal-Indicator-p-1383027.html?cur_warehouse=CN&rmmds=search)
With some PC tools like qModMaster (https://sourceforge.net/projects/qmodmaster/) you can directly can connect your PC to the ModBus device and can check if the module is working. 
[image]

This helps me to understand the whole things better...

emelianov, 2022-01-04:
@puneethintellithink 

It looks like TD321D485H uses inverse logic for data direction control. Try (note mb.begin()):

```c
void setup()
{
  Serial.begin(BAUD_RATE);
  Serial2.begin(BAUD_RATE, SERIAL_8E1);
  Serial.println();
  Serial.println("START");
  Serial.println();
  mb.begin(&Serial2, RS485_MODE, false);
  mb.master();
```

Also double check communication settings are identical on MDM and ESP sides.

closed 2022-01-13 by puneethintellithink