emelianov/modbus-esp8266 #177: numer of registers

joedupa, 2022-01-10:
Hi,
I can't pull more than 29 input registers in one readIreg() or readHreg() call. Is this a limitation of this library?

emelianov, 2022-01-11:
No limitation for read except available memory.

Can you explain what do you mean under "can't pull":
- `readHreg` returns `false`
- operation timeout
- devce crashes
- other behaviour

Also can you tell what device type (esp or other) and Modbus variation (TCP/RTU) do you using? What is slave/server device is queried?

Code example may also be useful to understand the issue reason.

joedupa, 2022-01-12:
I don't know if I can be technically eloquent here but will try.
I am trying to get values of the first 80 input registers from a RTU interface of the solar charge controller: Tristar MPPT 60 
( https://www.stellavolta.com/content/MSC-TS-MPPT-MODBUS-Doc.pdf ) My device is a NodeMCU v 0.9 ESP9266.

The code example:

void ModbusCall() {
    if (!mb.slave()) {
      mb.readIreg(1, 0, MB_BF1, 80, cbWrite);
   }
}
I can easily access/print the first 29 register values from this function but not the remaining ones, the value of anything above register 29 is always 0.  I can confirm that they are not 0 by changing the starting address to higher value.
Alternatively, I am also using a python Modbus library, and can access values from all 80 registers without any problem.
Hmmmm.... Please let me know what other information should I submit, hope you can help. Thanks!!

emelianov, 2022-01-12:
It's not about the library it's about SoftwareSerial. It's needed to be configured with larger buffer. Something like:
```c
SoftwareSerial S1;
//..
void setup() {
 S1.begin(19200, SWSERIAL_8N1, D1, D2, false, 256, 0);
/..
}
```

joedupa, 2022-01-12:
YES!!! that was it, the buffer did the trick. I don't know what is the size of the default buffer, but all 80 registers wouldn't "fit"
Thank again for your help!!

closed 2022-02-15 by emelianov