emelianov/modbus-esp8266 #179: RTU slave reading it's own registers

VitRuzicka, 2022-01-28:
Hi,
I have an almost working communication with your great library. On master (PC) I can easily request data from ESP32 (temp etc.), but, when I try to write holding register from PC, the value changes in the register but I cannot detect it from my program. I was hoping some callback function for receiving data from master exists, but I didn't find it. 
My goal is let's say reading temperature from master and turning on/off led based on register value on slave. I'm trying to read the register value like this:

[image]

But doesn't seem to change. The mb.task() is of course running in loop.
Thank you for reading this in advance.
Sincerely,
Vit

emelianov, 2022-01-30:
Hello,

It looks like [onSet callback](https://github.com/emelianov/modbus-esp8266/blob/master/examples/Callback/onSet/onSet.ino) is exactly what you need.

Also `mb.Hreg(ADDRRGBR)` should return current value stored in `ADDRRGBR` holding register.

VitRuzicka, 2022-01-30:
HI,
thank you very much, that's what I've been looking for. The register indeed changes, it's my code that caused it to not work properly.

closed 2022-01-30 by VitRuzicka