emelianov/modbus-esp8266 #182: Compatibility with OneWire?

blaisebarrette, 2022-02-06:
I have a mudbus RTU slave on a esp32

//--- RS485 ----//
    #define RS485_TX_pin 17
    #define RS485_RX_pin 16
    #define RS485_RE_DE_pin 4 
    #define SLAVE_ID 2
    
    Serial1.begin(115200, SERIAL_8N1, RS485_RX_pin, RS485_TX_pin);
    mb.begin(&Serial1, RS485_RE_DE_pin);
    mb.slave(SLAVE_ID);

It works fine but if I try to use OneWire on GPIO 22 it no longer works.

any suggestion?

thank you!

- Blaise

emelianov, 2022-02-14:
I'm afraid It's about OneWire library. Try to use latest Arduino/ESP32 version (probably even development).

blaisebarrette, 2022-02-14:
I use VSC and PlatformIO.
I think I saw an exemple for using these temperature sensors without the library. It's extensively more complexe but I guess it's my only choice.

Thanks for the quick reply!

closed 2022-03-06 by emelianov

blaisebarrette, 2022-03-18:
I found an alternative to OneWire. It’s OneWireNG [https://github.com/pstolarz/OneWireNg](https://github.com/pstolarz/OneWireNg) and I managed to get it working by putting Modbus and OneWire on separate cores of the Esp32.

Tomat7, 2022-03-19:
Used "regular" OneWire together with ModbusTCP (by modbus-esp8266 library) about 3 years without any issues. Both library working on "standard" Core 1. Are you using WiFi connection in your sketch?

blaisebarrette, 2022-03-19:
No but I M using modbus RTU.
I really tried all sorts of things to get them work together but as soon as I used the OneWire connection, Modbus would break.