emelianov/modbus-esp8266 #183: Tx and RX with stm32

simogaspa84, 2022-02-09:
Hi guys.. 

I was able to configure the esp32 as slave and receive messages from a stm32 mcu (master). 
I triggered a callback for a write function to a specific message. 
I am not able to see the message with the response  (reponse at the write message) coming from the ESp32 toward the STM32 mcu.

These are the configuration that work from master to slave  (STm32 -> RSP32) 

STM32

static void MX_USART6_UART_Init(void)
{

  /* USER CODE BEGIN USART6_Init 0 */

  /* USER CODE END USART6_Init 0 */

  /* USER CODE BEGIN USART6_Init 1 */

  /* USER CODE END USART6_Init 1 */
  huart6.Instance = USART6;
  huart6.Init.BaudRate = 19200;
  huart6.Init.WordLength = UART_WORDLENGTH_8B;
  huart6.Init.StopBits = UART_STOPBITS_1;
  huart6.Init.Parity = UART_PARITY_NONE;
  huart6.Init.Mode = UART_MODE_TX_RX;
  huart6.Init.HwFlowCtl = UART_HWCONTROL_NONE;
  huart6.Init.OverSampling = UART_OVERSAMPLING_16;
  if (HAL_UART_Init(&huart6) != HAL_OK)
  {
    Error_Handler();
  }
  /* USER CODE BEGIN USART6_Init 2 */
  __HAL_UART_ENABLE_IT(&huart6, UART_IT_RXNE);
  /* USER CODE END USART6_Init 2 */

}

ESP32

Serial2.begin(19200, SERIAL_8N1, RXD_PIN, TXD_PIN);

Thanks for your comments

emelianov, 2022-02-14:
1. If stm and esp boards connected directly by crossover RX/TX pins of via RS-485 transceivers?
2. For initial debug try to reduce line speed to 9600.

simogaspa84, 2022-02-15:
Hi Emelianov... I did several test.. Master to slave communication is working where the master is the stm32 chip and the slave is the esp32 running your lib. 
From what I understood the esp32 should reply to the master in an automatic way ?
Is it correct? 
Can you tell me how the lib builds the response for example to a write register message ? 
Thanks a lot

simogaspa84, 2022-02-15:
```
static void MX_USART6_UART_Init(void)
{

  /* USER CODE BEGIN USART6_Init 0 */

  /* USER CODE END USART6_Init 0 */

  /* USER CODE BEGIN USART6_Init 1 */

  /* USER CODE END USART6_Init 1 */
  huart6.Instance = USART6;
  huart6.Init.BaudRate = 19200;
  huart6.Init.WordLength = UART_WORDLENGTH_8B;
  huart6.Init.StopBits = UART_STOPBITS_1;
  huart6.Init.Parity = UART_PARITY_NONE;
  huart6.Init.Mode = UART_MODE_TX_RX;
  huart6.Init.HwFlowCtl = UART_HWCONTROL_NONE;
  huart6.Init.OverSampling = UART_OVERSAMPLING_16;
  if (HAL_UART_Init(&huart6) != HAL_OK)
  {
    Error_Handler();
  }
  /* USER CODE BEGIN USART6_Init 2 */
  __HAL_UART_ENABLE_IT(&huart6, UART_IT_RXNE);
  /* USER CODE END USART6_Init 2 */

}
```

emelianov, 2022-02-15:
The library worker should be periodically called from main loop (or from dedicated thread just keeping in mind that the library itself is not thread-safe).
```c
void loop() {
  mb.task();
  yield();
}
```
 task() checks Serial buffer for valid Modbus frame and sends back the response frame.
Common media-agnostic part of Modbus frame processing is defined in Modbus.h/Modbus.cpp. Take a look `slavePDU` method.

simogaspa84, 2022-02-15:
Thanks for your answer Emilianov... just few consideration.. 

I am working with a serial transceriver for RS485... 
So I need to set the direction of the data with a dedicated PIN (set by output)..
Usually It works only in receive mode and this is way I was able to only receive data.. 
So i need  once I have received a modbus message from the master to set the transceiver in tx mode.. and once the response is finished i have to set it back to the receive mode.. 
Is it possible to create a callback for triggering the end of the transmission of the slave ? 

I have tried to send from the master the following modbus message 

[image]

It should write one single register.. 

But the response from the lib and the esp32 is the following....  

[image]

It seems it took a Write Multiple Holding Registers on Modbus... 

What could be the issue ? 

Thanks a lot

emelianov, 2022-02-15:
> I need to set the direction of the data with a dedicated PIN (set by output)..

No problem. The library can handle direction pin internally.

```c
mb.begin(&Serial2, REDE_PIN);
```

> Is it possible to create a callback for triggering the end of the transmission of the slave ?

So such callback and manual direction control is not needed.

> It seems it took a Write Multiple Holding Registers on Modbus... 
> What could be the issue ?

Write single register is called if you passing single uint16_t by value to writeHreg:

```c
uint16_t value = 100;
mb.writeHreg(slave_id, REG, value); // (slave_id, REG, value, cb) if needed
```

simogaspa84, 2022-02-15:
Is there a way to print out the modbus response frame packet in the uart ? 
Thanks

emelianov, 2022-02-15:
> Does this command
> mb.begin(&Serial2, REDE_PIN);
> In automatic way changes the direction of the transmission ? Due to the fact that esp32 is a modbus slave it has to receive but then to transmit to sending the response.. Is it correct?

Sure. The library switches REDE as needed by itself.

> I transmit as modbus with a stm32 microcontroller.. In the picture above the first picture is the modbus tx message write single register with a default address 32 and transmitting the value 125.. I don't understand why the message that I get from the esp32 where the esp32 is the slave it is replying as it has received write multiple messages.. 
> Do you know what could be wrong ?

I've previously misunderstood you.
By now basing on data frames you provided before, it seems to me that the problem is in receiver (STM) side.

> Is there a way to print out the modbus response frame packet in the uart ?

Uncomment
```c
#define MODBUSRTU_DEBUG
```
in `ModbusSettings.h`
> Thanks

simogaspa84, 2022-02-16:
I did a bit of debug.. 
Transmitted from stm32 

[image]

emelianov, 2022-02-16:
Responce bytes is right payload for the request you are sending:

- [ 01 ] - not printed by debug but will be sent  - slave_id
- 03 - function code response for
- 02 - number of response data bytes
- 00 FF - responce data. Single register value in your case
- [00 00] - not printed by debug but will be sent - CRC

simogaspa84, 2022-02-17:
Thanks Emelianov.. 

So it seems correct from esp32 side.. 
I have to understand why the frame is wrong on the stm32 when it receive the response from the slave. 

Last 2 questions. 

How can I print in debug also the slave_id and CRC value computed ? 
How do you compute the CRC value ? 

Thanks again for your time

emelianov, 2022-02-17:
> How can I print in debug also the slave_id and CRC value computed ?
> How do you compute the CRC value ?

You may add extra debug output to `rawSend` in `ModbusRTU.cpp`
```c
bool ModbusRTUTemplate::rawSend(uint8_t slaveId, uint8_t* frame, uint8_t len) 
```

simogaspa84, 2022-02-17:
Thanks again.. 
I modified the print function and now I got 

[image]

At this point it seems to me a hw problem from the esp32 to the stm32... TX esp32 -> RX Stm32.. 
I will verify with oscilloscope

closed 2022-03-06 by emelianov