emelianov/modbus-esp8266 #184: How to query data

stritti, 2022-02-15:
Hi, I am very new using modbus.
I want to query data of a component, which I was able by CASModbusScanner:
[image]

How will this be done using modbus-esp8266?
I tried following Code but instead of value `21`  I receive `13792` on ESP.

I use following code:

```cpp
#include <Arduino.h>
#include <WiFi.h>
#include <ModbusIP_ESP8266.h>

WiFiClient  theClient;  // Set up a client for the WiFi connection
const char* ssid     = "SSID";
const char* password = "PWD";

IPAddress remote(192, 168, 178, 60);  // Address of Modbus Slave device
const int PORT     = 1502;
const int SLAVE_ID = 71;

ModbusIP mb;  //ModbusTCP object

bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) {  // Callback to monitor errors
  if (event != Modbus::EX_SUCCESS) {
    Serial.print("Request result: 0x");
    Serial.println(event, HEX);
  }
  return true;
}

void setup() {
  Serial.begin(74880);
  while (!Serial) {
  }

  Serial.println(F("Connecting to WiFi"));
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print(F("."));
  }

  IPAddress wIP = WiFi.localIP();
  Serial.printf("WIFi IP address: %u.%u.%u.%u\n", wIP[0], wIP[1], wIP[2], wIP[3]);

  // Set up ModbusTCP client.
  // mb.client();
  mb.slave(SLAVE_ID);

  Serial.println(F("Device started"));
}

#define REG_COUNT 2
const int32_t showDelay = 5000;  // Show result every n'th millisecond
uint32_t      showLast  = 0;

void loop() {

  if (millis() - showLast > showDelay) {  // Display register value every 5 seconds (with default settings)
    showLast = millis();
    uint16_t res[REG_COUNT];
    if (mb.isConnected(remote)) {                                     // Check if connection to Modbus Slave is established
      uint16_t trans = mb.readHreg(remote, 514, res, REG_COUNT, cb);  // Initiate Read Hreg from Modbus Server
      while (mb.isTransaction(trans)) {                               // Check if transaction is active
        mb.task();
        delay(10);
      }
    } else {
      mb.connect(remote, PORT);  // Try to connect if no connection
      Serial.println("Modbus connected.");
    }

    Serial.println(res[0]);
  }
}

```

emelianov, 2022-02-15:
To make request exactly as CASModbusScanner you need to specify `UnitId`:
```c
uint16_t trans = mb.readHreg(remote, 514, res, REG_COUNT, cb, SLAVE_ID); 
```
Otherwise standard `ignore` = 255 value is used.

stritti, 2022-02-15:
Awesome, @emelianov that was the missing point! Now it works. I searched for hours ...
Many thanks for this hint.

closed 2022-02-15 by stritti