emelianov/modbus-esp8266 #185: Wemos/Lolin D1 mini as modbus slave

gahujipo, 2022-02-19:
I am trying to use Wemos/Lolin D1 mini as Modbus slave to offer the master 4 registers. Is there a way to achieve this without having to use a MAX485 and by directly attaching the bus to a Pin of the ESP8266 or is MAX485 mandatory?

milen100, 2022-02-19:
@gahujipo You need some RS-485 converter/tranceiver - for example TTL/3.3V level to RS-485 . The RS-485 signal cannot be simulated with the ESP hardware only on software level.

gahujipo, 2022-02-19:
Thank you very, very much.

gahujipo, 2022-02-24:
I received some MAX485 chips a few days ago but I am not able to get the communication working. I used this example: 
https://github.com/emelianov/modbus-esp8266/blob/master/examples/RTU/slave/slave.ino

I can't find any information on how to wire the MAX485 to the Wemos D1: which PIN to use on the MAX465 side and on which on ESP side and where to define which pin I use on Wemos D1 side.

emelianov, 2022-02-25:
`DI` -> `RX_PIN`
`RO` -> `TX_PIN`
`RE`, `DE` (both) -> `REDE_PIN`
For example
`DI` -> `D4`
`RO` -> `D5`
`RE`, `DE` -> `D3`
```c
SoftwareSerial S(D4, D5, false, 256);
//...
  S.begin(9600, SWSERIAL_8N1);
  mb.begin(&S, D3);
```

gahujipo, 2022-02-25:
Thank you very much. That helps a lot. 
Which library would you use for SoftwareSerial. I found this (https://github.com/plerup/espsoftwareserial) one but it has no constructor with 4 parameters. Which one would you use in this example?

emelianov, 2022-02-27:
You can use just
```c
SoftwareSerial S(D4, D5);
```

gahujipo, 2022-03-11:
I just received today some MAX**3**485 (instead of MAX3485) tranceivers, because I read that this is needed for controllers with 3.3V levels which Wemods D1 apparently is. I connected it to the Wemos D1 mini with the following setup without any success: 

```DI``` -> ```D4```
```RO``` -> ```D5```
```RE```, ```DE``` (both) -> ```D3```

and this simple example code: 

```C++
#include <ModbusRTU.h>
#include <SoftwareSerial.h>

#define REGN 10
#define SLAVE_ID 1

ModbusRTU mb;
SoftwareSerial S(D4, D5);

void setup() {
  S.begin(9600, SWSERIAL_8N1);
  mb.begin(&S, D3);
  mb.setBaudrate(9600);

  mb.slave(SLAVE_ID);
  mb.addHreg(1,123);
  mb.Hreg(REGN, 100);
}

void loop() {
  mb.task();
  yield();
}
```

I tried to Read holding register(4x) 
Register number: 1 and 10
Data type: 16-bit unsigned integer
ModBus address: 1
Baud: 9600
Stopbits: 1
Parity: no parity

Have you a hint what to check next?

emelianov, 2022-03-15:
Reg 10 also should be added:
```c
  mb.addHreg(1,123);
  mb.addHreg(REGN, 100);
```

What is the error on master side? If it's getting timeout or some error code?

Next you may enable debug output on slave side to view incoming data. That may give ideas on what's wrong.
Uncomment in ModbusSettings.h
```c
#define MODBUSRTU_DEBUG
```

emelianov, 2022-03-27:
If no debug output at all try inverse direction logic: 
```c
mb.begin(&S, D3, false);
```
Then play with swapping RX/TX or/and A/B wires.

closed 2022-04-12 by emelianov