emelianov/modbus-esp8266 #186: Modbus master write single coil examples

legaxus-mis, 2022-02-20:
Dear Author,
Appreciate if you could offer some examples for Modbus Master Write Single Coil using ModbusRTU library, thanks.
Hope to hear from you soon. Thanks.

emelianov, 2022-02-20:
It's pretty much the same as read:
```c
bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) { // Callback to monitor errors
  if (event != Modbus::EX_SUCCESS) {
    Serial.print("Request result: 0x");
    Serial.print(event, HEX);
  }
  return true;
}
void loop() {
  bool coil  = true;
  if (!mb.slave()) {    // Check if no transaction in progress
    mb.writeCoil(SLAVE_ID, REG, coil, cb); // Send Write Coil request to Modbus Server
    while(mb.slave()) { // Check if transaction is active
      mb.task();
      delay(10);
    }
  }
  delay(1000);
}
```

legaxus-mis, 2022-02-21:
> It's pretty much the same as read:
>
> ```c
> bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) { // Callback to monitor errors
>   if (event != Modbus::EX_SUCCESS) { [...]

Hi Emelianov,
Thanks for your prompt reply, it works, thanks for the example.

closed 2022-03-06 by emelianov