emelianov/modbus-esp8266 #189: Dinamyc MODBUSRTU_MAX_READ_US

gonzabrusco, 2022-02-25:
Hello friend.

This is not an issue per se. It's just a recommendation based on what I saw with this library. Feel free to do what you think is best.

On `ModbusRTUTemplate::task()` you have one line that says:

`if (micros() - taskStart > MODBUSRTU_MAX_READ_US) { // Prevent from task() executed too long`

This is a good idea but also a bad idea, depending on how crowded is the 485 serial line. If we consider the maximum modbus frame length (256 bytes), a standard character size of 11 bits (1 start + 8 data + 2 stop), and a serial speed of for example 9600 bauds. A frame with 256bytes would need 256 * 11 / 9600 = 0.293 seconds. But you set your maximum time on 100 ms. That means that the read of one frame could be split in two calls to task(). 

The problem is that if the timeout occurs in the exact moment when the whitespaces was about to occur (that means, the frame was about to finish), then you could miss it! And if you are not the only slave on the bus, you can't miss anything! (because you could lose syncronization).

In my opinion, MODBUSRTU_MAX_READ_US should be calculated automatically with this formula:

`MODBUSRTU_MAX_READ_US  = 270 * 11 * 1000000 / baudrate`  (270 bytes just to be sure!)

That way, if you set a baudrate of for example 115200, then the maximum frame time un microseconds would be 25781 and but if you set a baudrate of 1200, it would be 2475000 (two seconds and a half!!).

The other way I think of making this library not blocking, is take advantage of the new Serial.onReceive() of the new Arduino 2.0.2 stack. Only call task() when you actually detect a whitespace, not before. But that could need a lot of reworking of the modbus RTU part.

emelianov, 2022-02-27:
Indeed
```c
if (micros() - taskStart > MODBUSRTU_MAX_READ_US) { // Prevent from task() executed too long
```
is just for what is written in comment: for prevent loop from locking. If the frame processing wasn't fit in single `MODBUSRTU_MAX_READ_US` period the rest of the frame will be buffered in by UART code and processed on next loop. So no frame lost is expected.
By now I'm not ready to change implementation of task() from general (suitable for most Arduinos) and ESP32-specific implementation.

gonzabrusco, 2022-02-27:
You could lose a frame, that is what I'm telling you. If you break the task() function on that point. When you return, you could have lost a "the whitespace" between two frames and continue buffering a response from another slave. That is what could happen in a busy RS485 network with the current code.

closed 2022-02-27 by gonzabrusco

emelianov, 2022-03-06:
@gonzabrusco 

> break the task() function on that point. When you return, you could have lost a "the whitespace" between two frames
In my opinion it's abnormal situation. task() must be called multiple times within Modbus timeout time. If loop duration may exceed this time amount loop should be refactored or split to multiple tasks (in case of ESP32). Probably it make sense to put Modbus code to dedicated task.

gonzabrusco, 2022-03-07:
Yes, I have it running in a separate task now. The only problem I see with that approach, if the baudrate is 9600 and the char size is  10 bits, then the interframe time would be 3.6 ms approx. That means that to detect that interframe time, the second task should run VERY fast and it could make starve other FreeRTOS tasks (minimum TICK time in ESP32 is 1 millisecond). Anyway, I don't see a "good" solution yet to this problem. I have to think about this a bit more.

emelianov, 2022-03-08:
@gonzabrusco 
Finally got you point. Yes, unfortunately slave may loose frames in case of intensive data exchange between master and multiple slaves on the bus.
The problem has no generic solution and may (or may not) be solved for specific device type by specific low-level way.
I'll put the issue into long term todo list.