emelianov/modbus-esp8266 #193: [TCP] Too many connections?

stritti, 2022-03-21:
Hi @emelianov 

I use your library to create an e-paper device to monitor solar power converter: https://github.com/stritti/kostal-pv-monitor

The device will go to deep sleep for 1min and queries via modbus data of converter to update the e-paper.
Now I have to see, that after some days the device does not get correct data any more. It looks like there are too many connections open or something like this. 

I know, this is very device specific, but probably I implemented your lib in wrong way. It would be awesome If you could spend a little time to review my example. Every suggestion is welcome.

emelianov, 2022-03-27:
Hello,
I don't think the issue is library related. For the sketch and the library every wake event looks just like starting from by power on.
Basing on ESP32 documentation the only one suggestion:

Try to use `esp_wifi_stop()` before go to deep sleep:
> Before entering deep sleep or light sleep modes, applications must disable WiFi and BT using appropriate calls ([esp_bluedroid_disable()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_bt_main.html#_CPPv421esp_bluedroid_disablev), [esp_bt_controller_disable()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/controller_vhci.html#_CPPv425esp_bt_controller_disablev), [esp_wifi_stop()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv413esp_wifi_stopv)).

stritti, 2022-03-27:
Hello @emelianov 

that is a good suggestion. I will try this in a week (not at home next week).

Thank you very much for your investigation and time invest.