emelianov/modbus-esp8266 #194: Question about multiple registers

simogaspa84, 2022-03-28:
Hi Emelianov... first of all thanks for your support.. 
I have a simple question about a callback function.. 

I have defined some registers in this way 

[image]

And then I have defined which callback to call in this way .. . 

[image]

As you can see every modbus register has from 1 to more registers inside (third parameter in the function def)

In case of  2 registers for the same modbus message How could I change the callback function for retrieving the values ? 

[image]

Thanks again for you support

emelianov, 2022-03-29:
Hello,
If you want to use the same callback function for multiple registers you can check register address on which callback is called. Refer [Use one callback function for multiple registers](https://github.com/emelianov/modbus-esp8266/blob/master/examples/Callback/onGetShared/onGetShared.ino) example.
Note that callback on read/write multiple registers Modbus request callback will be called for each affected register individually.

simogaspa84, 2022-03-29:
Hi Emelianov... I don't get the point.. 
In case I have defined with just one register as in this case  (NUM_REGISTERS_LOG = 1)

     **mb.onSetHreg(EVENTS, call_events,NUM_REGISTERS_LOG);**

I now that in the call back function in the val parameter I have the value inside the register 

**static uint16_t call_events(TRegister* reg, uint16_t val) 
{
  Serial.println(val);
  xQueueSend( logs_queue_from_uart_to_wifi, &val, portMAX_DELAY ==pdPASS);
  return val;
}**

But in case I have defined a register that can cointain 2 values (2 uint16_t)  

**mb.onSetHreg(ALARMS, call_alarm,NUM_REGISTERS_ALARMS);**   (NUM_REGISTERS_ALARMS = 2)

How can I change the call back function for getting the first and the second parameter ? 

**static uint16_t call_alarm(TRegister* reg, uint16_t val) 
{
  ?????
  return val;
}**

Thanks a lot for your time and help

emelianov, 2022-03-29:
Let's imagine `onSetHreg` is used as
```c
mb.onSetHreg(10, callback_func, 2)
```
After this the same `callback_func` will be called on setting Hreg 10 and Hreg 11 (zero based counting).
As Modbus operates only 16-bit values you need to process high and low parts of 32-bit value depending on register offset call-back called against.
Refer issue #158 for good explanation of such data processing.

closed 2022-04-12 by emelianov