emelianov/modbus-esp8266 #195: Limitation on the amount of writeHreg to 3 register

bernardojchw, 2022-03-28:
On  my slave side:

32 hreg   waitting to recive data on a mega 2560 that is busy updating  a 4" screen.

on my master side:

arduino UNO  tcp server and RTU master

rtu.writeHreg(1, _i , _reg, 3, cb);
more then 3 will get a 00xE4 error (timeout) i had to divide in blocos of 3
void slave1Update(){
  uint16_t _reg[3];
    for (uint16_t  _i = UC; _i < TOTAL_REGS_SIZE; _i=_i+3) {
      for (uint16_t  _v= 0;_v <4 ; _v++){
      _reg[_v] = au16data[_i+_v];
      }
    if (!rtu.slave()) {
        rtu.writeHreg(1, _i , _reg, 3, cb);

        while(rtu.slave()) { // Check if transaction is active
            rtu.task();
            delay(1000);
         }
         like this is working
         
         but i have more 4 slaves to read and write do you have a solution?

I whish to write 32 Hreg on one comand.

emelianov, 2022-03-29:
In general I see no issues in you code preventing from reading more than 3 registers. I suspect larger data read/write request become failed as serial loosing synchronisation because of base oscillator frequency drift or SoftwareSerial inability to provide stable bit flow. All you can try to do without replacing hardware is to test different SoftwareSerial versions.

emelianov, 2022-04-02:
In general HardwareSerilal is more reliable but as far as I understand you are using Uno with ingle Serial that is busy for debug peruse.
As for SoftwareSerial no suggestion on specific version. I just noticed that some versions working better with specific hardware. That is sometimes latest version is not the best one for specific situation.

closed 2022-04-17 by emelianov