emelianov/modbus-esp8266 #196: Can't write to slave hreg

RosiersRobin, 2022-03-29:
Hi,

I try to change the modbus address of a slave, but it doesn't seem to do anything..

This is how I'm doing it;

```
modBusSerial.begin(9600, SWSERIAL_8E1, MB_RX, MB_TX); // modbus configuration SWSERIAL_8E1 = 8 bits data, even parity and 1 stop-bit

    if (!modBusSerial) {
        Serial.println("ModBus serial (software serial) is not configured correctly.");
    }

    modbus.begin(&modBusSerial, MB_TR, true);
    modbus.master();

void writeModbusHoldingRegister(uint8_t address, uint16_t offset, uint16_t value) {
    if (!modbus.slave()) {
        modbus.writeHreg(address, offset, value, modBusCallBack);
    }

    // process
    while (modbus.slave()) {
        modbus.task(); // processes the data
        vTaskDelay(1);
    }
}

writeModbusHoldingRegister(16, 106, 22);
```

So the above code I want to change address 16 (stored in holding register 106) to be changed to 22. but nothing hapens after it and address 16 remains active.

am I missing something here?

The modBusCallBack is a global defined callback that would report if the action failed, but it doesn't.

emelianov, 2022-03-29:
In case if callback reported EX_SUCCESS result code data had successfully sent to slave. So everything is okay with master and physical connection.
I have only two ideas:
1. Slave documentation uses 1-based register addressing (the library is 0-based) and you need to write value to register 105.
2. The device needs to be restarted to activate changes.

RosiersRobin, 2022-03-30:
Soo, I'm an idiot...

The writing works as it should, but the modbus slave doesn't do anything with it....

I can write the slave the new value in it's holding register, but then the slave says "meh I just keep using my old address"...

Thanks @emelianov for the insights tho! Really appriciate it!

closed 2022-03-30 by RosiersRobin