emelianov/modbus-esp8266 #197: [Help wanted] Write raw value or write register using addresses

RosiersRobin, 2022-03-30:
Hello, me again 😅 

So I have 2 commands;
`01 10 00 57 00 02 04 00 01 00 00 E6 B5`
`01 10 00 57 00 02 04 00 01 00 01 27 75`

These work when I throw them in modbus poll, but if I understand them correctly, they write to two registers at the same time?

In the examples folder, I see that there is a function `writeMultiple(SLAVE_ID, HREG(HREG_ID), 10);` , but I don't really understand that one....

I would like call twice 
`modbus.writeHreg(address, offset, value);` (I have no idea what values I should throw in there....)

per single command, but I don't think that's how it should be done...

Any idea's? 🙏

RosiersRobin, 2022-03-31:
Update on this;

I have this working in modpoll like this;

[image]

But I see that's using function code 16 instead of 06, any idea on how to achieve this?

emelianov, 2022-04-02:
> they write to two registers at the same time?

The library uses the same function name for `FC_WRITE_REG = 0x06` and `FC_WRITE_REGS  = 0x10` functions. Specific function is selected based on arguments passed:

```c
uint16_t values[2] = {1, 2};
mb.writeHreg(SLAVE_ID, HREG_ID, (uint16_t)VALUE); // Single uint16_t value passed
mb.writeHreg(SLAVE_ID, HREG_ID, (uint16_t*)values, 2); // Pointer to uint16_t array and array size passed
```

> Can I send a binary too?

Just start number with `0b`. 
```c
writeModbusHoldingRegister(1, 864, 0b0110010000000011);
```

RosiersRobin, 2022-04-02:
That `0b` seems to be the correct thing to do here, thanks!

However, as mentioned in the screen with that "2 registers", I actually ment the 0x16 function code to be executed. Since I need to modify a register using code 16 (as showen in the screenshot)

emelianov, 2022-04-03:
This works for single value as well:
```c
uint16_t value = 100;
mb.writeHreg(SLAVE_ID, HREG_ID, (uint16_t*)value, 1); // Pointer to uint16_t
```

RosiersRobin, 2022-04-03:
That doesn't seem to be working, it should use function code 16, but it still uses function code 06

this is what I do;

```cpp
void writeModbusHoldingRegister(uint8_t address, uint16_t offset, uint16_t value) {
    if (!modbus.slave()) {
        modbus.writeHreg(address, offset, value); // callback does not work (yet) here
    }

    // process
    while (modbus.slave()) {
        modbus.task(); // processes the data
        vTaskDelay(1);
    }
}

uint16_t off = 0;
uint16_t on = 1;
while (true) {
    Serial.println("Turning on relay....");
    writeModbusHoldingRegister(1, 88, on);
    delay(5000);
    Serial.println("Turning off relay....");
    writeModbusHoldingRegister(1, 88, off);
    delay(5000);
}
```

RosiersRobin, 2022-04-05:
Allright, so that worked already, just the binary thing doesn't work (I think)... I'll investigate that one further and let you know!

RosiersRobin, 2022-04-06:
Nevermind the last comment, it worked, the modbus device was just really slow on showing it... thanks ALOT for the help and keep up the AMAZING work!!

closed 2022-04-06 by RosiersRobin