emelianov/modbus-esp8266 #200: Вопрос по применению библиотечных функций.

PabloFed, 2022-04-12:
Здравствуйте Александр! Возникла проблема записи в память ПЛК частотного преобразователя.
`if (!mb.slave()) {           
    mb.readHreg(SLAVE_ID, READ_REG, res, REG_COUNT, cb);    
    while(mb.slave()) {      
      mb.task();        
      delay(50);
    }
    Serial.println(res[0]);    
  }`
Конструкция такого плана успешно функционирует. Регистры читаются. 

`if (!mb.slave()) {    
      mb.writeHreg(SLAVE_ID, READ_REG, res, REG_COUNT, cb); 
      while(mb.slave()) { 
        mb.task();
        delay(10);
      }
    }`
Конструкция такого плана значения в регистры не пишет.
Использую ESP32 devkitV1 как master. А частотный преобразователь как slave. Call back последней конструкции - Request result: 0x5. Подскажите если есть мысли. Спасибо.

PabloFed, 2022-04-13:
Настроил в частотнике формат данных как в вашей библиотеке. 8N1 (была 8N2) и без проверки четности. Ни на что не повлияло. С добавлением вышеуказанных правок в мониторе порта ESP32 наблюдаем следующее:
RTU Slave: 1, Fn: 03, len: 4, RTU Slave: 1, Fn: 90, len: 2, Request result: 0x5
Я так понимаю первая часть это запрос от мастера на чтение (напомню что чтение работает без проблем). А Что есть вторая часть после первой len:4?

PabloFed, 2022-04-13:
Что есть Fn:90?

PabloFed, 2022-04-13:
#define MODBUSRTU_TIMEOUT 1000 - менял на разные значения. Без результата.

PabloFed, 2022-04-13:
#include <ModbusRTU.h>
//#include <ESP8266_LCD_1602_RUS.h>
//#include <Keypad.h>

//////////   MODBUS   /////////////
#define SLAVE_ID 1            //номер сервера
#define READ_REG 3            //читаемый регистр
#define WRITE_REG 5       //запиываемый регистр
#define REG_COUNT 1           //количество читаемых регистров
#define MODBUS_RX_PIN 16      //пин RX
#define MODBUS_TX_PIN 17      //пин TX
#define MODBUS_DE_RE 4        //пин DE_RE

ModbusRTU mb;

bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) {   
  if (event != Modbus::EX_SUCCESS) {
    Serial.print("Request result: 0x");
    Serial.println(event, HEX);
  }
  return true;
}
uint16_t res[REG_COUNT]; 

void setup() {
  Serial.begin(115200);         
  Serial2.begin(9600, SERIAL_8N1, MODBUS_RX_PIN, MODBUS_TX_PIN);    //запускаем и определяем параметры MODBUS
  mb.begin(&Serial2, MODBUS_DE_RE);     //работа по состоянию пина 4
  mb.master();
}

void loop() {
        ////////////////////////////////  READ  ///////////////////////////////////////
  if (!mb.slave()) {           
    mb.readHreg(SLAVE_ID, READ_REG, res, REG_COUNT, cb);    
    while(mb.slave()) {      
      mb.task();        
      delay(50);
    }   
  }
  Serial.println(res[0]);
  delay(1000);
    
      //////////////////////////// WRITE ///////////////////////////////////////////
     if (!mb.slave()) {    
      mb.writeHreg(SLAVE_ID, WRITE_REG, res, REG_COUNT, cb); 
      while(mb.slave()) { 
        mb.task();
        delay(50);
      }
    }
  
}

PabloFed, 2022-04-13:
Может что - то в коде. Необходимо содержимое регистра 3 записать в регистр 5. Оба регистра для чтения/записи.

PabloFed, 2022-04-13:
У частотника 5 настроек связи по RS485 там запутаться невозможно. Адрес на шине (ID), скорость передачи, формат передачи данных (RTU, ASCII), формат данных (8 - N - 2, 8 - N - 1), проверка четности или нет.

emelianov, 2022-04-13:
0x90 = 0x80 + 0x10 -- Error response на функцию 16.
Он как-бы не совсем error -- принято к исполнению.
Обмен данными выполняется корректно. Почему параметры не записываются в регистры -- вопрос к частотнику. Надо курить его мануалы или к разработчикам обращаться.

PabloFed, 2022-04-14:
Правильно ли я понимаю что в строке:
mb.writeHreg(SLAVE_ID, WRITE_REG, res, REG_COUNT, cb); 
при условии что   uint16_t res[REG_COUNT] = {18}, а  REG_COUNT = 1,
значение res (то бишь 18) для  передачи преобразуется в шестнадцатиричное восьмибитное число?

PabloFed, 2022-04-14:
Просто я взял частотник иного производителя и с ним CallBack на запись не ругается, но, всё равно, ничего не пишет. Вот я и предположил, что где - то в передаваемых данных проблема.

PabloFed, 2022-04-14:
`#include <ModbusRTU.h>
//#include <ESP8266_LCD_1602_RUS.h>
//#include <Keypad.h>

//////////   MODBUS   /////////////
#define SLAVE_ID 1            //номер сервера
#define READ_REG 8468            //читаемый регистр
#define WRITE_REG 8192       //запиываемый регистр
#define REG_COUNT 1           //количество читаемых регистров
#define MODBUS_RX_PIN 16      //пин RX
#define MODBUS_TX_PIN 17      //пин TX
#define MODBUS_DE_RE 4        //пин DE_RE

ModbusRTU mb;

bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) {   
  if (event != Modbus::EX_SUCCESS) {
    Serial.print("Request result: 0x");
    Serial.println(event, HEX);
  }
  return true;
}
uint16_t res[REG_COUNT];
uint16_t pusk[REG_COUNT] = {18};

Modbus::ResultCode cbRtuRaw(uint8_t* data, uint8_t len, void* custom) {
  auto src = (Modbus::frame_arg_t*) custom; // <custom> argument contains some header data on incoming packet
  Serial.printf("RTU Slave: %d, Fn: %02X, len: %d, ", src->slaveId, data[0], len);
  if ((Modbus::ResultCode)data[0] ==  Modbus::EX_ACKNOWLEDGE)   // Check if incoming is Acknowledge
    return Modbus::EX_ACKNOWLEDGE;  // Instruct the library to stop the packet processing
  return Modbus::EX_PASSTHROUGH;      // or process packet normally
}
 

void setup() {
  Serial.begin(115200);         
  Serial2.begin(9600, SERIAL_8N1, MODBUS_RX_PIN, MODBUS_TX_PIN);    //запускаем и определяем параметры MODBUS
  mb.begin(&Serial2, MODBUS_DE_RE);     //работа по состоянию пина 4
  mb.master();
  mb.onRaw(cbRtuRaw);
}

void loop() {
        ////////////////////////////////  READ  ///////////////////////////////////////
  if (!mb.slave()) {           
    mb.readHreg(SLAVE_ID, READ_REG, res, REG_COUNT, cb);    
    while(mb.slave()) {      
      mb.task();        
      delay(50);
    }   
  }
  Serial.println(res[0]);
  delay(1000);
    
      //////////////////////////// WRITE ///////////////////////////////////////////
     if (res[0] == 1){

     if (!mb.slave()) {    
      mb.writeHreg(SLAVE_ID, WRITE_REG, pusk, REG_COUNT, cb); 
      while(mb.slave()) { 
        mb.task();
        delay(50);
      }
    }
    Serial.println(pusk[0]);
  }
}`

PabloFed, 2022-04-14:
Блок READ читает состояние регистра 8468 (нажата ли кнопка).
Блок WRITE выполняется если  в регистре 8468 появляется 1 (кнопка нажата).
Условие отрабатывается так как в монитор порта ESP32 выводится значение pusk[0].
Но вот в регистр 8192 это значение не пишется.

PabloFed, 2022-04-14:
RTU Slave: 1, Fn: 03, len: 4, 0
RTU Slave: 1, Fn: 03, len: 4, 0
RTU Slave: 1, Fn: 03, len: 4, 0
Это периодически выводится если кнопка не нажата.

RTU Slave: 1, Fn: 03, len: 4, 1
RTU Slave: 1, Fn: 10, len: 5, 18
Это если кнопка нажата.

PabloFed, 2022-04-14:
Частотник исправен. С иных, конечно же специализированных устройств, ввод значений в регистры PLC частотника происходит корректно.

PabloFed, 2022-04-14:
По спецификации на частотники  - код функции записи 06. Если я правильно понимаю при записи тут Fn: 10.

PabloFed, 2022-04-14:
Может частотники не адекватно реагируют на функцию записи нескольких регистров?

PabloFed, 2022-04-14:
Мои терминалы для общения с частотниками, кстати, и не умеют в 0x10. Есть возможность Вашими средствами попробовать 0x06?

emelianov, 2022-04-15:
Если передать `writeHreg` значение uint16_t, а не ссылку на массив, то будет использована функция 06:
```c
mb.writeHreg(SLAVE_ID, WRITE_REG, pusk[0], cb); 
```

closed 2022-05-02 by emelianov

PabloFed, 2022-05-05:
Прошу прощения за долгий перерыв.
mb.writeHreg(SLAVE_ID, WRITE_REG, pusk[0], cb); 
Такое указание на первый элемент массива не работает.

no matching function for call to 'ModbusRTU::writeHreg(int, int, uint16_t&, int, bool (&)(Modbus::ResultCode, uint16_t, void*))'

PabloFed, 2022-05-05:
uint16_t pusk[REG_COUNT] = {18};
Вот так объявлен масссив с присвоением значения переменной.
Можно ли вообще передавать не массив, а значение одной переменной?

PabloFed, 2022-05-05:
uint16_t pusk = 18;
.........
mb.writeHreg(SLAVE_ID, WRITE_REG, pusk, REG_COUNT, cb);
Такое тоже не компилируется.

PabloFed, 2022-05-05:
Понял. Количество регистров не нужно передавать!

PabloFed, 2022-05-05:
Низкий Вам поклон за участие!