emelianov/modbus-esp8266 #201: Modbus TCP client постоянный таймаут.

pavellzubkov, 2022-04-16:
Приветствую, Александр!
Подскажите пожалуйста почему при попытке использовать readHreg из клиента постоянно получаю Modbus::EX_TIMEOUT.
Код запускаю на ESP32 в задаче FreeRtos . В качестве сервера для теста использую [программу](http://www.ardsoft.ru/mEmulator.html).  Соединение с сервером устанавливается, сервер получает запрос и сразу отправляет ответ, но почему-то этот ответ не принимается. 
В трассировке эмулятора сервера вижу , что он получает запрос - 05 4E 00 00 00 06 01 03 00 C9 00 01   и сразу отправляет ответ 05 4E 00 00 00 05 01 03 02 00 00   и вроде все должно быть хорошо, но не хорошо)
пробовал сервер заводить тоже с вашей библиотеки на ESP8266 но результат тот же. Ниже код (для теста тут надерганы куски из примеров.. в результате в main.cpp в setup  я запускаю ModbusNet::Run(&mainState);  где mainState это указатель на мою глобальную структуру там со всякими штуками, но она тут явно непричем. Да, таймаут согласно настройкам 1 секунда , что так и происходит.

```
#include <Arduino.h>
#include <stdlib.h>
#include <inttypes.h>
#include "ModbusStartClass.h"

#ifdef ESP8266
#include <ESP8266WiFi.h>
#else
#include <WiFi.h>
#endif
#include <ModbusTCP.h>

const int REG = 201;                // Modbus Hreg Offset
IPAddress remote(192, 168, 1, 139); // Address of Modbus Slave device

ModbusTCP mb; // ModbusTCP object

namespace ModbusNet
{    namespace
    {
        bool cb(Modbus::ResultCode event, uint16_t transactionId, void *data)
        {                                                      // Modbus Transaction callback
            if (event != Modbus::EX_SUCCESS)                   // If transaction got an error
                Serial.printf("Modbus result: %02X\n", event); // Display Modbus error code
            if (event == Modbus::EX_TIMEOUT)
            {                          // If Transaction timeout took place
                mb.disconnect(remote); // Close connection to slave and
                mb.dropTransactions(); // Cancel all waiting transactions
            }
            return true;
        }

        void taskReadModbus(void *pvParameters)
        {
            AppState *appState;
            appState = (AppState *)pvParameters;
            Serial.println(" Modbus task runing create");
            // uint8_t increment = 0;
            uint16_t res = 2;
            while (true)
            {
                bitflip(appState->display.Leds, 2);
                if (mb.isConnected(remote))
                {
                    bitSet(appState->display.Leds, 1);
                    // Check if connection to Modbus Slave is established
                    uint16_t trans = mb.readHreg(remote, REG, &res, 1, cb, 1); // Initiate Read Hreg from Modbus Server
                    while (mb.isTransaction(trans))
                    { // Check if transaction is active
                        mb.task();
                        vTaskDelay(10 / portTICK_RATE_MS);
                    }
                    bitflip(appState->display.Leds, 2);
                    Serial.printf("Remote register ");
                    Serial.println(REG);
                    Serial.println(res); // At this point res is filled with responce value
                }
                else
                {
                    bitClear(appState->display.Leds, 1);
                    Serial.println("try connect modbus server");
                    mb.connect(remote); // Try to connect if no connection
                }
                // delay(100);

                vTaskDelay(100 / portTICK_RATE_MS);
            }
        };
    }

    void Run(AppState *state)
    {
        mb.client();
        xTaskCreatePinnedToCore(
            taskReadModbus, /* Функция задачи */
            "ModbusTask",   /* Название задачи */
            10000,          /* Размер стека задачи */
            (void *)state,  /* Параметр задачи */
            1,              /* Приоритет задачи */
            NULL,           /* Идентификатор задачи,
                                 чтобы ее можно было отслеживать */
            1);             /* Ядро для выполнения задачи (1) */
    };
}
```

emelianov, 2022-04-17:
В последнем коммите был поломан transaction id. Fixed.
6600ea5

pavellzubkov, 2022-04-17:
О, спасибо! отлично, а то я уже два дня нервничал не понимал где затык)

closed 2022-04-17 by pavellzubkov