emelianov/modbus-esp8266 #203: MB RTU slave, Timeout error with baud rate > 19200

MarioGG7, 2022-04-22:
Hi.
Thanks for so powerful library

Spec: ESP32 WROOM,  ST3485EB (3.3V),  Switchable termination resistor 120R, Arduino IDE, Scan rate time = 1S.

Library 4.0.0:

I use Mb RTU with around 100  write/read registers  and I don't have any issue with baud rate <= 19200.
Higher  baud rate cause the timeout error.

To eliminate the bug from my project, I created simple  Arduino code with single  Modbus register based on your example, and problem is exactly the same.

Can you suggest something please?
Increase the scan rate time  to 2s didn't help.

After when I updated the Library to  4.1.0 I have:
- Timeout error
- Read error: framing error
- Write error

My simply code:

/*
  ModbusRTU ESP8266/ESP32
  Simple slave example

  (c)2019 Alexander Emelianov (a.m.emelianov@gmail.com)
  https://github.com/emelianov/modbus-esp8266

  modified 13 May 2020
  by brainelectronics

  This code is licensed under the BSD New License. See LICENSE.txt for more info.
*/

#include <ModbusRTU.h>
#include <SoftwareSerial.h>
#define REGN 0

#define SLAVE_ID 1

#define MB_Enabled_pin 5
#define MB_RX 16  // D7
#define MB_TX 17  // D6

SoftwareSerial RS485Serial_S(MB_RX, MB_TX);
ModbusRTU mb;

//#define SSerialTxControl 5

void setup() {
// <= 19200  ALL WORKS FINE // > 19200 Timeout error
  RS485Serial_S.begin(38400, SWSERIAL_8N2);
  mb.begin(&RS485Serial_S, MB_Enabled_pin);  //or use RX/TX direction control pin (if required)
  mb.setBaudrate(38400);
  mb.slave(SLAVE_ID);
  mb.addHreg(REGN);

  mb.Hreg(REGN, 100);

}

void loop() {
  mb.task();
  yield();
}

emelianov, 2022-04-23:
It may be SoftwareSerial or transceiver problem.

Suggestions for SoftwareSerial issue:
1. Use HardwareSerial if possible (ESP32 has 3 of them)
2. Try different versions of SoftwareSerial library (probably not the latest one may work)

Suggestions transceiver issue:
1. Try to use another power source and/or protect power lines by additional capacitors.
2. Replace transceiver(s)

MarioGG7, 2022-04-25:
Hi Alexander

Thank you for fast response.

Problem as usually was between keyboard and chair :-) and with Software Serial. 
When I use Hardware Serial, than all is fine > 19200.
I was lacky, because I choose USART 2 pins on ESP32 in my project.

Thank you.

closed 2022-05-09 by emelianov