emelianov/modbus-esp8266 #206: unable to read any slave values as master without 2 task()'s

RobWPink, 2022-04-29:
currently trying to make a function that attempts to connect to all slaves within a range, then list those slaves that are connected in an array. However, without 2 task() functions (one before and one after the readCoil) I cant get any values to print. Even with the two task()'s I end up getting the wrong numbers (since its probably reading the previous data in the buffer). Also the cbWrite doesn't print out on some cycles, not sure why that is.

Slave code seems to work fine and is using this: to create the coils `mb.addCoil(0, true, 1);`

For reference, this function is meant to be called once in setup(), and I have 3 arduino megas each has their own MAX485 chip hooked up to hardware Serial1. DE/RE are hooked up to pin 2 as denoted in the code by `mb.begin(&Serial1,2);`

Any help on this would be immensely appreciated. I would normally ask stackOverflow, but this is more of a library question.

```
bool cbWrite(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  Serial.print("Request result: 0x");
  Serial.print(event, HEX);
  Serial.println("");
  return true;
}

void checkSlaves(ModbusRTU mbNum, uint8_t *_availableSlaves, int maxSlaves){
  bool res;
  int cnt = 0;
  Serial.println("Searching for slaves...");
  for(uint8_t i = 0; i < maxSlaves; i++){
    for(int j = 0; j < 10; j++){
      mbNum.task();
      mbNum.readCoil(i, 0, &res, 1, cbWrite);
      mbNum.task();
      yield();
      if(res){
        Serial.print("Slave #");
        Serial.print(i);
        Serial.println(" found!");
        _availableSlaves[cnt] = i;
        cnt++;
        res = false;
        break;
      }
      delay(100);
    }
  }
  Serial.println("Search complete");
}
```

Output:
Slave ID's 3 and 4:
```
Searching for slaves...
Slave #3 found!
Slave #4 found!
Slave #5 found!
Search complete
```
Slave ID's 2 and 5:
```
Searching for slaves...
Slave #2 found!
Slave #3 found!
Slave #5 found!
Slave #6 found!
Search complete
```

emelianov, 2022-04-30:
```c
//...
for(int j = 0; j < 10; j++){
      mbNum.readCoil(i, 0, &res, 1, cbWrite);
      while(mbNum.slave()) {
          mbNum.task();
          yield();
      }
      //...
```

RobWPink, 2022-05-02:
holy crap that worked, THANK YOU! it looks so simple so I was surprised! As thanks if you need anything 3D printed I'd be happy to help!

closed 2022-05-15 by emelianov