emelianov/modbus-esp8266 #209: Maximum number of registers handled by the library

simogaspa84, 2022-05-09:
Hi  ..

I would like to know how many modbus registers the library can handle.. 
For example If i send a write register command with 100 registers can the library handle it ?

Thanks a lot 

Simone

emelianov, 2022-05-09:
Hello Simone,

Read/Write registers count is limited by ModbusRTU frame size (256 bytes - header/crc).
So up to 125 registers can be processed within single request.

simogaspa84, 2022-05-09:
Thanks emelianov.. .Is there also a max baudrate that the lib can support ?  
Thanks again

emelianov, 2022-05-09:
> Thanks emelianov.. .Is there also a max baudrate that the lib can support ?

It's hardware depended I think.
In my tests HardwareSerial ESP32 (master) to ESP32 (slave) direct link was running with no packet last up to 1Mbps.