emelianov/modbus-esp8266 #211: Modbus TCP communication ENC28J60 and STM32

elektroboard, 2022-05-25:
Hello. I want to communicate ModbusTCP with ESP32 and ENC28j60 ethernet module. However, I couldn't. What sample code should I use?

emelianov, 2022-05-25:
Hi.
[ModbusEthernet](https://github.com/emelianov/modbus-esp8266/tree/master/examples/TCP-Ethernet) should work for ENC28J60 too.

elektroboard, 2022-05-26:
Hi.
I try to your "server.ino" file. . But there is a error and dont connect. An error code appeared: Trying to connect to remote TCP server...
Socket error #10061 . What'is problem?

emelianov, 2022-05-27:
It's [reported](https://github.com/emelianov/modbus-esp8266/issues/101#issuecomment-752189158) that the library is working with enc28j60 using [UIPEthernet library](https://github.com/UIPEthernet/UIPEthernet).
So yon need to modify example code to use UIPEthernet.h instead Ethernet.h.

elektroboard, 2022-05-27:
Thank. I will try.

closed 2022-06-11 by emelianov

JAlvares20, 2024-10-11:
Hello. Next week I'll do the same project with the same components. I just want to know if what you did worked?

emelianov, 2024-10-27:
No any news since than.