emelianov/modbus-esp8266 #212: ModbusTCP with ESP32

pkrssy, 2022-06-06:
Hi Emelianov,

I use your ModbusRTU library to read data from my heat pump controller with success and I also want to connect to my solar inverter with ESP32 using your example script TCP-ESP/client.ino, 
The register value is reported to be zero although it works without problem with QModMaster. The example script is:

`/*
  Modbus-Arduino Example - Master Modbus IP Client (ESP8266/ESP32)
  Read Holding Register from Server device

  (c)2018 Alexander Emelianov (a.m.emelianov@gmail.com)
  https://github.com/emelianov/modbus-esp8266
*/

#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>

const int REG = 32085;               // Modbus Hreg Offset
IPAddress remote(192, 168,88, 90);  // Address of Modbus Slave device
const int LOOP_COUNT = 10;

ModbusIP mb;  //ModbusIP object

void setup() {
  Serial.begin(115200);
 
  WiFi.begin("xxxx", "vvvvvv");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  mb.client();
}

uint16_t res = 0;
uint8_t show = LOOP_COUNT;

void loop() {
  if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
    mb.readHreg(remote, REG, &res);  // Initiate Read Coil from Modbus Slave
  } else {
    mb.connect(remote, 502);           // Try to connect if no connection
  }
  mb.task();                      // Common local Modbus task
  delay(100);                     // Pulling interval
  if (!show--) {                   // Display Slave register value one time per second (with default settings)
    Serial.println(res);
    show = LOOP_COUNT;
  }
}`

Output of serial: 

 11:11:28.081 -> .....
11:11:30.107 -> WiFi connected
11:11:30.107 -> IP address: 
11:11:30.107 -> 192.168.88.205
11:11:31.236 -> 0
11:11:32.399 -> 0
11:11:33.562 -> 0
11:11:34.692 -> 0
11:11:35.889 -> 0
11:11:37.019 -> 0
11:11:38.149 -> 0

[image]

Can you please help what I am doing wrong?

emelianov, 2022-06-07:
Basing on you screenshot it looks like UnitId should be passed to `readHreg`:
```c
#define PUMP_UNIT_ID 1
//...
mb.readHreg(remote, REG, &res, 1 nullptr, PUMP_UNIT_ID);
```
(the library default for UnitId is 255 = don't care)

pkrssy, 2022-06-07:
Issue solved. Thank you.

closed 2022-06-25 by emelianov