emelianov/modbus-esp8266 #214: Ethernet Modbus Client with Modbus server (PLC device)

tarekanwer, 2022-06-14:
Hi there, 
I am currently working on an esp32 dev board with ethernet shield module W5100. I am trying to get data from certain holding registers from a PLC device (looking at it as a server). I used the following device getting series of 2s on the serial monitor. Hope you can help. Thanks in advance.

`/*
  ModbusTCP for W5x00 Ethernet library
  Basic Client code example

  (c)2020 Alexander Emelianov (a.m.emelianov@gmail.com)
  https://github.com/emelianov/modbus-esp8266

  This code is licensed under the BSD New License. See LICENSE.txt for more info.
*/

#include <SPI.h>
#include <Ethernet.h>       // Ethernet library v2 is required
#include <ModbusEthernet.h>

const uint16_t REG =2025;               // Modbus Hreg Offset
IPAddress remote(192, 168, 1, 1);  // Address of Modbus Slave device
const int32_t showDelay = 5000;   // Show result every n'th mellisecond

// Enter a MAC address and IP address for your controller below.
byte mac[] = {
  0x00, 0x1A, 0xD3, 0x40, 0x9A, 0xED
};
IPAddress ip(192, 168, 137, 1); // The IP address will be dependent on your local network:
ModbusEthernet mb;               // Declare ModbusTCP instance

void setup() {
  Serial.begin(115200);     // Open serial communications and wait for port to open
  #if defined(AVR_LEONARDO)
  while (!Serial) {}        // wait for serial port to connect. Needed for Leonardo only
  #endif
  Ethernet.init(5);         // SS pin
  Ethernet.begin(mac, ip);  // start the Ethernet connection
  delay(1000);              // give the Ethernet shield a second to initialize
  mb.client();              // Act as Modbus TCP server
}

uint16_t res = 2;
uint32_t showLast = 0;

void loop() {
if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
    mb.readHreg(remote, REG, &res);  // Initiate Read Hreg from Modbus Slave
  } else {
    mb.connect(remote);           // Try to connect if not connected
  }
  delay(100);                     // Pulling interval
  mb.task();                      // Common local Modbus task
  if (millis() - showLast > showDelay) { // Display register value every 5 seconds (with default settings)
    showLast = millis();
    Serial.println(res);

  }
}`
[image]
I am expecting at the serial 100 instead of 2.

emelianov, 2022-06-14:
Generic TCP/IP diagnostic is suggested first: verify that both client and server ip available by ping. Probably you need to specify network mask and gateway in ethernet init.
Check if ESP's client is able to connect to server. Use code based on Transactional callback example which provides information of the client-server request result (error code).

tarekanwer, 2022-06-15:
it was solved when i assigned the rest of parameters in the readHreg function as follows : 
`mb.readHreg(remote, REG, &res, 1 ,nullptr, 1);`