emelianov/modbus-esp8266 #216: Continuously read and write data through MODBUS TCP

FelipeMPaulino, 2022-06-22:
Hi Emelianov, 
First of all hope to find you well and I also thank you for your work.

 I am trying to use your library in an application with a Siemens PLC (1200) as master and an ESP32 + W5500 eth board as slave on Modbus TCP.
 I am having trouble understanding how it works to continuously read and write on the registers/coils inside the void loop( ), once I am now having connection issues from the ESP32 side after the first read or write. After the first read or write I can not repeat a transaction before restarting the ESP32.
  May you please share a example code to continuously read/write Modbus registers/coils?
  
  I have tried many codes, one of them is:
  
  
////////////////////////////////////////////////////////////////////////////////
const int reg1_HREG = 0;
const int reg2_HREG = 1;
const int reg3_HREG = 2;
const int reg10_HREG = 10;
const int reg11_HREG = 11;
const int reg12_HREG = 12;

byte mac[] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};

IPAddress localip(192, 168, 20, 2);
IPAddress plcmaster(192,168,20,1);
IPAddress gateway(192, 168, 20, 15);
IPAddress subnet(255, 255, 255, 0);

ModbusEthernet mb;

void setup() {

  Serial.begin(115200);
  Ethernet.init(33); //ESP32 with Adafruit Featherwing Ethernet
  Ethernet.begin(mac, localip);  //start the Ethernet connection
  delay(1000); //give the Ethernet shield a second to initialize

 if (Ethernet.hardwareStatus() == EthernetNoHardware) {
    Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");  
    while (true) {
        delay(1); // do nothing, no point running without Ethernet hardware
    }  }
 if (Ethernet.linkStatus() == LinkOFF) {
    Serial.println("Ethernet cable is not connected.");}

    mb.server(); //Act as Modbus TCP server

    mb.addHreg(reg1_HREG, 21);
    mb.addHreg(reg2_HREG, 07);
    mb.addHreg(reg3_HREG, 93);
    
    mb.addHreg(reg10_HREG, 1);
    mb.addHreg(reg11_HREG, 1);
    mb.addHreg(reg12_HREG, 1);
}

void loop() {

   Serial.print("HR 10: ");
   Serial.print(mb.Hreg(reg10_HREG));
   Serial.print(" , HR 11: ");
   Serial.print(mb.Hreg(reg11_HREG));
   Serial.print(" , HR 12: ");
   Serial.println(mb.Hreg(reg12_HREG));

   int teste1 = random(1,4095);
   mb.addHreg(reg1_HREG, teste1);

   mb.task(); //Server Modbus TCP queries

  yield();
  delay(1000);
}
////////////////////////////////////////////////////////////

tks in advanced!

franciscogimeno2000, 2022-06-22:
Hello, a question how do you intend to change the value of a constant?
Check the code.

Fran.

FelipeMPaulino, 2022-06-22:
Sorry, you are right. Choose the wrong code to share. I shall use a "mb.writeHReg" for example. 
Unfortunately, even sctruring the code with this function I still have the issue, no more then one transmition and the ESP needs to be reseted to get back reading/writing. 
In the code I have shared, there is also a mb.Hreg. Isnt this one meant to read the registers value? The main idea was to read always.

franciscogimeno2000, 2022-06-22:
This estructure is the next one:

In the setup the registers are defined and in the loop it reads or writes them, but it does not define them again with mb.addHreg in the loop.

closed 2022-07-06 by emelianov