emelianov/modbus-esp8266 #221: Trying to compile code for Portenta H7 - Using Ethernet and not WiFi

salasidis, 2022-06-30:
I compiled the example code

``
/*
  ModbusTCP for W5x00 Ethernet library
  Basic Server code example

  (c)2020 Alexander Emelianov (a.m.emelianov@gmail.com)
  https://github.com/emelianov/modbus-esp8266

  This code is licensed under the BSD New License. See LICENSE.txt for more info.
*/

#include <SPI.h>
#include <Portenta_Ethernet.h>
#include <Ethernet.h>       // Ethernet library v2 is required
#include <ModbusEthernet.h>

// Enter a MAC address and IP address for your controller below.
byte mac[] = {
  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED
};
IPAddress ip(192, 168, 30, 177); // The IP address will be dependent on your local network:
ModbusEthernet mb;              // Declare ModbusTCP instance

void setup() {
  Serial.begin(115200);     // Open serial communications and wait for port to open
  #if defined(AVR_LEONARDO)
  while (!Serial) {}        // wait for serial port to connect. Needed for Leonardo only
  #endif
  Ethernet.init(5);        // SS pin
  Ethernet.begin(mac, ip);  // start the Ethernet connection
  delay(1000);              // give the Ethernet shield a second to initialize
  mb.server();              // Act as Modbus TCP server
  mb.addReg(HREG(100));     // Add Holding register #100
}

void loop() {
  mb.task();                // Server Modbus TCP queries
  delay(50);
}
``

The only change was adding the line
``
#include <Portenta_Ethernet.h>
``

On the top

I get error messages at compile time

``

In file included from C:\Users\Robert_Salasidis.SALASOFTMAIN\Documents\Arduino\libraries\modbus-esp8266\src/ModbusEthernet.h:12:0,
                 from C:\Users\RS\AppData\Local\Temp\arduino_modified_sketch_918696\server.ino:14:
C:\Users\RS\Documents\Arduino\libraries\modbus-esp8266\src/ModbusTCPTemplate.h: In instantiation of 'void ModbusTCPTemplate<SERVER, CLIENT>::task() [with SERVER = EthernetServerWrapper; CLIENT = arduino::EthernetClient]':
C:\Users\RS\AppData\Local\Temp\arduino_modified_sketch_918696\server.ino:36:11:   required from here
C:\Users\RS\Documents\Arduino\libraries\modbus-esp8266\src/ModbusTCPTemplate.h:217:72: error: 'class EthernetServerWrapper' has no member named 'accept'
   while (millis() - taskStart < MODBUSIP_MAX_READMS && (c = tcpserver->accept())) {
                                                             ~~~~~~~~~~~^~~~~~
In file included from C:\Users\RS\Documents\Arduino\libraries\modbus-esp8266\src/Modbus.h:16:0,
                 from C:\Users\RS\Documents\Arduino\libraries\modbus-esp8266\src/ModbusAPI.h:8,
                 from C:\Users\RS\Documents\Arduino\libraries\modbus-esp8266\src/ModbusEthernet.h:11,
                 from C:\Users\RS\AppData\Local\Temp\arduino_modified_sketch_918696\server.ino:14:
C:\Users\RS\Documents\Arduino\libraries\modbus-esp8266\src/darray.h:45:10: error: 'size_t DArray<T, SIZE, INCREMENT>::find(UnaryPredicate, size_t) [with UnaryPredicate = ModbusTCPTemplate<SERVER, CLIENT>::task() [with SERVER = EthernetServerWrapper; CLIENT = arduino::EthernetClient]::<lambda(TTransaction&)>; T = TTransaction; int SIZE = 2; int INCREMENT = 2; size_t = unsigned int]', declared using local type 'ModbusTCPTemplate<SERVER, CLIENT>::task() [with SERVER = EthernetServerWrapper; CLIENT = arduino::EthernetClient]::<lambda(TTransaction&)>', is used but never defined [-fpermissive]
   size_t find(UnaryPredicate func, size_t i = 0) {
          ^~~~
exit status 1
Error compiling for board Arduino Portenta H7 (M7 core).
``

Is there a known issue compiling for the Portenta? Any suggestions to resolve this.

Thanks

emelianov, 2022-06-30:
The code have updated to be compatible with Portenta in [d8bc667](https://github.com/emelianov/modbus-esp8266/commit/d8bc667b614aa2938485df32ef1e986277b4fd18)
Please test with code from master branch. I was't able to test it with real hardware but it should work now.

salasidis, 2022-06-30:
I commented out the Ethernet.init(5) line, and it compiled OK - not tested, but at least compiles

salasidis, 2022-07-01:
Unfortunately, I get the equivalent of the Blue screen of death on the portents - (the mBed OS crashes).

It happens on the mb.task() routine in the code below - the serial.print - prints the entering mb.task, but not the exiting

(I added a longer delay to get the ethernet connection stablle)

``
/*
  ModbusTCP for W5x00 Ethernet library
  Basic Server code example

  (c)2020 Alexander Emelianov (a.m.emelianov@gmail.com)
  https://github.com/emelianov/modbus-esp8266

  This code is licensed under the BSD New License. See LICENSE.txt for more info.
*/

#include <SPI.h>
#include <Portenta_Ethernet.h>
#include <Ethernet.h>       // Ethernet library v2 is required
#include <ModbusEthernet.h>

// Enter a MAC address and IP address for your controller below.
byte mac[] = {
  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED
};
IPAddress ip(192, 168, 0, 202); // The IP address will be dependent on your local network:
ModbusEthernet mb;              // Declare ModbusTCP instance

void setup() {
  Serial.begin(115200);     // Open serial communications and wait for port to open
  #if defined(AVR_LEONARDO)
  while (!Serial) {}        // wait for serial port to connect. Needed for Leonardo only
  #endif
  //Ethernet.init(5);        // SS pin
            Serial.println("Starting Ethernet");
  Ethernet.begin(mac, ip);  // start the Ethernet connection
  delay(10000);              // give the Ethernet shield a second to initialize
          Serial.println("Entering mb.server");
  mb.server();              // Act as Modbus TCP server
          Serial.println("Exiting mb.server");
  mb.addReg(HREG(100));     // Add Holding register #100
}

void loop() {
        Serial.println("Entering mb.task");
  mb.task();                // Server Modbus TCP queries
        Serial.println("Exiting mb.task");
  delay(50);
}
``

I can try to debug the mb.task routine on my hardware if you can suggest where to look for possible issues.

Thanks

salasidis, 2022-07-01:
I made these changes to the library - just to see where it is crashing

``
	
	if (tcpserver) {
		CLIENT c;
		// WiFiServer.available() == Ethernet.accept() and should wrapped to get code to be compatible with Ethernet library (See ModbusTCP.h code).
		// WiFiServer.available() != Ethernet.available() internally
		
			Serial.println("in tcpserver");
		
		while (millis() - taskStart < MODBUSIP_MAX_READMS && (c = tcpserver->accept())) {
				Serial.println("in accepted");
#if defined(MODBUSIP_DEBUG)
			Serial.println("IP: Accepted");
#endif
			CLIENT* currentClient = new CLIENT(c);
			if (!currentClient || !currentClient->connected())
				continue;
#if defined(MODBUSRTU_DEBUG)
			Serial.println("IP: Connected");
#endif
			if (cbConnect == nullptr || cbConnect(currentClient->remoteIP())) {
				#if defined(MODBUSIP_UNIQUE_CLIENTS)
				// Disconnect previous connection from same IP if present
				n = getMaster(currentClient->remoteIP());
				if (n != -1) {
					tcpclient[n]->flush();
					delete tcpclient[n];
					tcpclient[n] = nullptr;
				}
				#endif
				n = getFreeClient();
				if (n > -1) {
					tcpclient[n] = currentClient;
					BIT_SET(tcpServerConnection, n);
#if defined(MODBUSIP_DEBUG)
					Serial.print("IP: Conn ");
					Serial.println(n);
#endif
					continue; // while
				}
			}
			// Close connection if callback returns false or MODBUSIP_MAX_CLIENTS reached
			delete currentClient;
		}
	}
	
	Serial.println("post accept");
	``

This is the output

`Post Ethernet
Entering mb.server
Exiting mb.server
Entering mb.task
in mb.task
post cleanup
in tcpserver
`

It seems to crash on the line 

	while (millis() - taskStart < MODBUSIP_MAX_READMS && (c = tcpserver->accept())) {

It does not get to "in accepted" or "post accepted"

salasidis, 2022-07-01:
The problem is the assignment of c = tcpserver->accept();

`	if (tcpserver) {
		CLIENT c;
		// WiFiServer.available() == Ethernet.accept() and should wrapped to get code to be compatible with Ethernet library (See ModbusTCP.h code).
		// WiFiServer.available() != Ethernet.available() internally
		
			Serial.println("in tcpserver");
			
			c = tcpserver->accept();
			
			Serial.println("post blank accept");
		
		while (millis() - taskStart < MODBUSIP_MAX_READMS && (c = tcpserver->accept())) {`

If I write c = tcpserver->accept(); above, it crashes right at that statement.

If I write  tcpserver->accept();  (without the assignment to c), it does not crash until the wh8ile statement that follows

Hope that helps

emelianov, 2022-07-01:
It looks like my mistake.
Pushed updated code which I believe will finally get it working.

salasidis, 2022-07-01:
I loaded the new  ModbusEthernet.h file (as well as ghoing back to the original TCPTemplate file.

Still MBed crassh (blicking SOS on Portenta).

I went backj to my modified Teplate file (with all the Serrial.print statements) and it is still crashing at the same place  - where the assignmewnt to 

			c = tcpserver->accept();

occurs.

emelianov, 2022-07-06:
Have you tested basic ethernet server examples with Portenta? I'm not sure that Portenta's Ethernet library server code is consistent.

salasidis, 2022-07-06:
I am using the ethernet as a web server on the same device. That works OK.

I used the ArduinoModubus (the other modbus library), and that works OK, with no modifications.

Robert

emelianov, 2022-07-08:
I have no ideas on what is going wrong and see no way to diagnose problem without having real board for debug.

salasidis, 2022-07-08:
If there was a remote debugging feature,  I could  help.

salasidis, 2022-07-13:
Does your version of Server TCP allow multiple clients at once.

The ArduinoModbus only seems to support one client connection at a time, and trying to make it work with more seems to be very cumbersome, and gives a lot of read errors or timeout errors when more than one is connected

Thanks

emelianov, 2022-07-15:
> Does your version of Server TCP allow multiple clients at once.

Sure. The library allows multiple client/server instances running on the device. Just note that active TCP connection count is limited by hardware/software to 4-8.

I hope to have spare time to provide debug library build to diagnose problem with Portenta next week if problem still relevant.

salasidis, 2022-07-15:
There is an AsyncWebServer that I use for the webserver side of things, and that allows as many browsers to connect as required, and all the Ajax requests are handled well.

Maybe that would work for the modbus side as well?

emelianov, 2022-07-15:
> There is an AsyncWebServer that I use for the webserver side of things, and that allows as many browsers to connect as required, and all the Ajax requests are handled well. Maybe that would work for the modbus side as well?

AsyncTCP usage is in long term library's todo list.

Web-server clients slightly differ from Modbus. Typically web-client connects, reads page/image/file and disconnects. Modbus client connects and stays connected for a days. I mean that simultaneous persistent connections count are limited.

salasidis, 2022-07-15:
There is a Portenta_H7_AsyncTCP library as well – it was used as the base to the AsyncWebServer library

The way I would see the library working is that it would spawn threads if a new connection were available, and drop connections (except the base one) if a connection was idle for a period of time. This may presumably cause heap fragmentation issues. The other way would be to simply say that x number of simultaneous clients are supported, and just start that many threads at the startup (likely easier to do the latter).

emelianov, 2022-07-23:
Please try https://github.com/emelianov/modbus-esp8266/tree/M7debug branch.
TCPEthernet/server.ino example and the library core slightly changed. Can you test if the changes takes any effect to the issue?

salasidis, 2022-07-23:
Will do next week.

Thanks

closed 2023-01-22 by emelianov