emelianov/modbus-esp8266 #227: ModbusTCP to ModbusRTU bridge with on-device ModbusRTU simulator

javiergentoo, 2022-07-21:
Hi
Before, thank you very much for the contribution.

I have using this example to connect with a real RTU slave and I ask from tcp modbus.
I can see than when I send a request from tcp modbus, it go thrown the RTU Modbus and the device answer from Serial port ok.
but the "tcp.rawResponce(srcIp, data, len, slaveRunning);" do not send the data back to the TCP.
I have done a for into the data and the information is there, but I don't receive any into the tcp request.

I don't use the sym inside the example, I am using the Serial1 connecting to the slave RTU.

The RTU is working because I am receiving the request from tcp modbus and go on to RTU modbus. But in the PC I receive empty.

Into ESP32:

- TCP IP: 192.168.1.155 Fn: 04, len: 5
This is the for of the data:
4
0
5
0
12
120

And I receive this order message:

Modbus result: E4, Mem: 270660

What could I do wrong ?

javiergentoo, 2022-07-21:
Part of the code:

  if (src->unitId) {
    tcp.setTransactionId(transRunning); // Set transaction id as per incoming request

    uint16_t succeed = tcp.rawResponce(src->ipaddr, data, len, slaveRunning);

    for(int i = 0; i <= len; i++) {
      Serial.print(i);
      Serial.print(": ");
      Serial.println(data[i]);   // I have add this for watch the data and the data are coming perfect
    }

output from real slave:
TCP IP: 192.168.1.155 Fn: 04, len: 5 
0: 4
1: 0
2: 5
3: 0
4: 10
5: 120
fail:0   ----> this output come from function "Modbus::ResultCode cbRtuRaw(uint8_t* data, uint8_t len, void* custom)
RTU Slave: 1, Fn: 04, len: 22, Response TCP IP: 192.168.1.155

But I received an array empty in the PC:

emelianov, 2022-07-22:
Take a look fixed TCP-to-RTU-Simulator example changes I've just pushed.
Also note that response to TCP client is send from cbRtuRaw not from cbTcpRaw.

javiergentoo, 2022-07-22:
Hi Alexander
Thank you very much for your contribution, now it is working perfect.
only it has a error in compilation because it has a extra parenthesis in line 61:

-before: tcp.errorResponce(IPAddress(**(**src->ipaddr), (Modbus::FunctionCode)data[0], Modbus::EX_SLAVE_DEVICE_BUSY);

-after: tcp.errorResponce(IPAddress(src->ipaddr), (Modbus::FunctionCode)data[0], Modbus::EX_SLAVE_DEVICE_BUSY);

But it is working perfect.
Thanks

javiergentoo, 2022-07-22:
Hi Alexander

I have add this two lines and know it is working:

  if (transRunning) { // Note that we can't process new requests from TCP-side while waiting for responce from RTU-side.
    tcp.setTransactionId(src->transactionId); // Set transaction id as per incoming request
    tcp.errorResponce(IPAddress(src->ipaddr), (Modbus::FunctionCode)data[0], Modbus::EX_SLAVE_DEVICE_BUSY);
    **transRunning = 0;**
    **slaveRunning = 0;**
    return Modbus::EX_SLAVE_DEVICE_BUSY;
  }

emelianov, 2022-07-23:
I've updated the example code with slightly more correct solution than provided by your (but your direction was right).

javiergentoo, 2022-07-23:
Thank you very much for your help

javiergentoo, 2022-07-26:
Hi Emelianov

I have changed the example like your modification, but it doesn't work.
When I request an address where it is not present and return again to the address present the tcp is disconnect.

**With this lines works perfect:**

if (transRunning) { // Note that we can't process new requests from TCP-side while waiting for responce from RTU-side.
    tcp.setTransactionId(src->transactionId); // Set transaction id as per incoming request
    **tcp.errorResponce(IPAddress(src->ipaddr), (Modbus::FunctionCode)data[0], Modbus::EX_SLAVE_DEVICE_BUSY);
    transRunning = 0;
    slaveRunning = 0;**
    return Modbus::EX_SLAVE_DEVICE_BUSY;
  }

closed 2022-09-07 by emelianov