emelianov/modbus-esp8266 #229: ModbusTCP Slave modify the Slave address

ShyamBhattacharyya, 2022-07-22:
Hi Sir.
I want to develop ModbusTCP Slave program on multiple ESP8266 devices. So I need to change the Slave ID for each ModbusTCP slaves connected in the same TCP network. But I could not find anywhere in the documentation how to change the unitId of slave. For ModbusRTU slave it is 

> md.slave(unitId);

But how to do that for ModbusTCP slave?
For example my code is like the following(Slightly modified from your example of the Modbus Slave).
``` c
/*
  Modbus-Arduino Example - Test Holding Register (Modbus IP ESP8266)
  Configure Holding Register (offset 100) with initial value 0xABCD
  You can get or set this holding register
  Original library
  Copyright by André Sarmento Barbosa
  http://github.com/andresarmento/modbus-arduino
  Current version
  (c)2017 Alexander Emelianov (a.m.emelianov@gmail.com)
  https://github.com/emelianov/modbus-esp8266
*/

#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else //ESP32
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>

// Modbus Registers Offsets
const int TEST_HREG = 100;
const uint16_t port = 502;
const uint16_t unitId = 5;                **// Please notice here**

//ModbusIP object
ModbusIP mb;

**//In here I want to declare a specific slave address for each slave, like**
mb.slave(unitId);                            **// Please notice here**
  
void setup() {
  Serial.begin(115200);
 
  WiFi.begin("your_ssid", "your_password");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  mb.server(port);
  mb.addHreg(TEST_HREG, 0xABCD);
}
 
void loop() {
   //Call once inside loop() - all magic here
   mb.task();
   delay(10);
}`
```

Tomat7, 2022-07-22:
On my opinion, the easiest way to separate multiply ModbusTCP slave devices is to separate by IP-address, not ID.

ShyamBhattacharyya, 2022-07-22:
Thank you for your valuable comment. It is helpful. But in my case, each esp8266 is connected with five different slaves in a daisy chain. I need to change the obtained data(it is not in modbus) of each slaves to modbus data of respective slaves. So that the master can see these devices like normal slaves connected to a specific IP having different IDs for five different slaves attached under each ESP8266.

emelianov, 2022-07-22:
The library architecture allows to implement such functionality only using `rawCallback`. General idea is the same as in [TCP-to-RTU example](https://github.com/emelianov/modbus-esp8266/blob/master/examples/Bridge/TCP-to-RTU-Simulator/TCP-to-RTU-Simulator.ino).

ShyamBhattacharyya, 2022-07-27:
Thank you sir for your response. Can you write me a sample? I could not understand the code.

closed 2023-01-01 by emelianov