emelianov/modbus-esp8266 #230: esp32 with MAX485

tarekanwer, 2022-07-23:
Hi there , 

Hope you are doing well. I need some help in using the library while using MAX485 module as it is not clear for me how to use it in this case. Thanks in advance

Regards,

emelianov, 2022-07-23:
Nothing special for MAX485. You may just add direction control pin definition:
```c
mb.begin(&Serial, REDE_PIN);
```

Merdock1, 2022-07-26:
Hello as data.
The modules that I use are of this type:
[It's just an example](https://www.amazon.com/ACEIRMC-Adapter-Converter-Indicator-Arduino/dp/B093348BQR?pd_rd_w=PgoGZ&content-id=amzn1.sym.bc622850-a717-4d94-96c3-7cc183488298&pf_rd_p=bc622850-a717-4d94-96c3-7cc183488298&pf_rd_r=0TA33FZW3J9FSNT6PK0Z&pd_rd_wg=C7kyl&pd_rd_r=ee9b5b24-bded-4c75-8382-df621ef0d12f&pd_rd_i=B093348BQR&psc=1&ref_=pd_bap_d_rp_13_i)
This type of module does not need a pin for flow control.
I hope it is useful for you.
Greetings.

vantuilpinheiro, 2022-08-02:
Hi, @tarekanwer. I'm also with doubts about it, but I got it to work with ESP32. 
The only issue I've been getting error messages 1/3 when reading Hregs.

I coded an example to test. It may help you. Look here: https://github.com/emelianov/modbus-esp8266/issues/233#issue-1325131736

closed 2022-08-25 by emelianov