emelianov/modbus-esp8266 #231: Multiple client connection on TCP server

DavideTurra, 2022-07-25:
thank you for this useful library. 

Is it possible to have multiple client connection for a TCP server in some way?

thanks in advance

emelianov, 2022-07-27:
Sure. Number of simultaneous connections for client or server is limited by Arduino's tcp stack implementation.
The only limitation that the library is support only single client (outgoing) connection per server IP.

DavideTurra, 2022-07-28:
ok thank you for the answer, may be I made some mistake somewere in my code. All the best to you

closed 2022-08-14 by emelianov

papercodeIN, 2024-03-09:
> Sure. Number of simultaneous connections for client or server is limited by Arduino's tcp stack implementation. The only limitation that the library is support only single client (outgoing) connection per server IP.

Means from different system with different IP address I can connect to Same Modbus TCP Server. So, this is how it will support multiple connection correct ?