emelianov/modbus-esp8266 #233: Modbus RTU Implementation - ESP32 (Master and Slave)

vantuilpinheiro, 2022-08-02:
Hi, @emelianov!
First of all, thank you for your contribution, attention, and great work with the library.

**Context:**
Lately, I've been working on the development of an ESP32 master-slave communication via Modbus RTU using your library. However, once I started to work with a higher data transmission frequency, I faced one problem. I get 1 error in the read function (Hreg) in every 3 samples collected. I simplified the problem to try to find the reason, but I'm unable to find it.

**Simplified Test:**
Slave: It acquires LDR measured values and sends them to the master when requested.
Master: Send the request to read the measured values by the slave.

[image]

//****----------------------//
**Slave Code:**
#include <ModbusRTU.h>

#define RXD2 16 //RO
#define TXD2 17 //DI
#define DIR 15 //RE e DE
#define REGN 0
#define SLAVE_ID 1
int ldrPin = 13;

ModbusRTU mb;

void setup() 
{
  pinMode(ldrPin, INPUT);
  
  Serial.begin(9600, SERIAL_8N1);
  Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2);
  mb.begin(&Serial2,DIR);

  mb.slave(SLAVE_ID);//Initializing modbus slave device with ID 1

  mb.addHreg(REGN); // add the register with address 0
  mb.Hreg(REGN, 99); 
}

void loop() 
{
  uint16_t ldrValor = analogRead(ldrPin);
  
  Serial.print("Valor LDR: ");
  Serial.println(ldrValor);
  mb.Hreg(REGN, ldrValor); 

  mb.task();
  yield();
}

//****----------------------//
**Master Code:**
#include <ModbusRTU.h>
#define RXD2 16 //RO
#define TXD2 17 //DI
#define DIR 15 //RE e DE

#define SLAVE_ID1 1
#define FIRST_REG 0
#define REG_COUNT 13

//variáveis globais para as temporizações
unsigned long tempo1 = 0;
unsigned long tempoAnterior1 = 0;

ModbusRTU mb;

uint16_t storing_data = 0; // place holder to store data

bool cbWrite(Modbus::ResultCode event, uint16_t transactionId, void* data) 
{
  //Identifica se houve erro
  if(event != Modbus::EX_SUCCESS) 
  {
    Serial.printf_P("Request result: 0x%02X, Mem: %d\n", event, ESP.getFreeHeap());
  }
  return true;
}

void setup() 
{
  Serial.begin(115200);
  Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2);
  mb.begin(&Serial2,DIR);
  mb.master();
}

void loop() 
{
  tempo1 = millis();

  if (tempo1 - tempoAnterior1 > 1000) 
  {
    tempoAnterior1 = millis();

    // Verifica se não existe transação em progresso (Se o canal esta livre)
    if (!mb.slave()) 
    {
      mb.readHreg(SLAVE_ID1, FIRST_REG, &storing_data, REG_COUNT, cbWrite);
      
      // Verifica se a transação está ativa
      while (mb.slave()) 
      { 
        mb.task();
        yield();
      }
      
      if (storing_data != 0)
      {
        Serial.print("Valor medido: ");
        Serial.println(storing_data);
      }
    } 
  }
}

emelianov, 2022-08-02:
There are two ideas:
- On some Arduito/ESP32 versions changing Serial2 to Serial3 may help.
- Reduce AnalogRead rate. That do not read it every loop cycle.

vantuilpinheiro, 2022-08-09:
Thanks for the attention, emelianov! 
I've never thought it could be wiring due to the pattern of error messages, but I checked the circuit. There was a mal functional cable. The problem was solved when I replaced it.

closed 2022-08-09 by vantuilpinheiro