emelianov/modbus-esp8266 #238: modbusRTU doesn't work with a specific device

pkrssy, 2022-08-14:
I am trying to connect to a power meter using ESP32 and a MAX485 interface modul but on serial monitor I see the script below stucks after a few seconds doing mb.task(). If Modbus comm. wires are disconnected, it works normal, timeout message repeats periodically.

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
Mb task
--- stucks here --

ESP32 is not frozen at that state as it answers to ping request if it is connected to wifi.
Script works well with another device on the bus. 
How can I debug it?

`#include <ModbusRTU.h>

uint32_t checktime;

uint16_t rec_data;
bool flag;

#define RXD2 16   //  -> RO 16
#define TXD2 17   //  -> DI 17
#define DIR 15    //  -> DE/RE 15

ModbusRTU mb;

bool cbWrite(Modbus::ResultCode event, uint16_t transactionId, void* data) {

  Serial.printf_P("Request result: 0x%02X, Mem: %d\n", event, ESP.getFreeHeap());
  flag = true;
  return true;
}

void setup() {
  Serial.begin(115200);
   Serial2.begin(9600, SERIAL_8E1, RXD2, TXD2);
  mb.begin(&Serial2, DIR);
  mb.master();

}

void loop() {
  
  if ( millis() > checktime )  
  {
    
    checktime = millis() + 5000; 
    
    if (!mb.slave()) {
      //mb.readHreg(1, 16388, &rec_data, 1, cbWrite);
      mb.readHreg(2, 1024, &rec_data, 1, cbWrite);
      while (!flag) 
      {
        delay(50);
        Serial.println("Mb task");
        mb.task();
      }
      flag = false;
      Serial.println(rec_data);
    }
  }
}`

emelianov, 2022-08-14:
It's not clear what could go wrong.
Try to enable debug output by uncommenting
```c
#define MODBUSRTU_DEBUG
```
in `ModbusSettings.h`.

Also it's recommended more generic way to wait for slave response:
```c
if (!mb.slave()) {
  //mb.readHreg(1, 16388, &rec_data, 1, cbWrite);
  mb.readHreg(2, 1024, &rec_data, 1, cbWrite);
  while (mb.slave())  // Do while transaction is active 
  {
    delay(50);
    Serial.println("Mb task");
    mb.task();
  }
```

emelianov, 2022-08-18:
Basing on you last log it's clear that slave device doesn't answer (the library 0xE4 error code).
Quick test on lab hardware goes as expected: request - timeout in 1 second - delay rest of 10 seconds - new  request.
Suggested actions: check wiring and power source quality.

pkrssy, 2022-08-18:
You missed my point: if you check serial log with my comment, after the Ttmeout message While loop is running for 8 more cycles then it stucks, controller needs to be reset (done after 1 minute and 17 seconds). While loop seems to be hung although ESP32 is responsive (I tried it was accessible from wifi). I should have here a timeout response after 10 seconds at 18:02:32:

...
18:02:22.717 -> while mb_slave
18:02:22.717 -> while mb_slave
18:02:22.750 -> while mb_slave
18:02:22.750 -> Request result: 0xE4, Mem: 375552                   <<<<<<<< PREVIOUS TIMEOUT
18:02:22.750 -> 0
18:02:31.434 -> 3 40 4 0 1
18:02:31.467 -> while mb_slave
18:02:31.467 -> while mb_slave
18:02:31.467 -> while mb_slave
18:02:31.467 -> while mb_slave
18:02:31.500 -> while mb_slave
18:02:31.500 -> while mb_slave
18:02:31.500 -> while mb_slave
18:02:31.500 -> while mb_slave               <<<<<<< THIS IS THE LAST LINE IN SERIAL LOG
18:03:48.768 -> ets Jun 8 2016 00:22:57  <<<<<< as more than 10 sec elapsed without activity, reset button is pressed here
18:03:48.768 ->
18:03:48.768 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
18:03:48.801 -> configsip: 0, SPIWP:0xee
18:03:48.801 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
18:03:48.801 -> mode:DIO, clock div:1
18:03:48.801 -> load:0x3fff0018,len:4

emelianov, 2022-08-19:
I do note that output is hang in your case. I just said that unable to reproduce this behaviour with me environment.
One more idea: if you are using Arduino under macOS it's probably just serial port processing problem on computer side. I've faced with random loosing debug output for some boards before.

pkrssy, 2022-08-21:
I use Arduino IDE under Linux (SuSE Leap 15.3)

closed 2022-09-23 by emelianov