emelianov/modbus-esp8266 #240: write single coil to trigger imput on robot with ESP32. ModBus TCP protocol

kiksonix, 2022-08-24:
I need help. I want to write single coil to trigger imput on robot. Coil adress is 1002. [01002]

I get E4 and E6 error.  

Code is bellow:

#ifdef ESP8266
#include <ESP8266WiFi.h>
#else
#include <WiFi.h>
#endif
#include <ModbusIP_ESP.h>

const int REG = 1002; // Modbus Coils Offset
const int COUNT = 1; // Count of Coils
IPAddress remote(172, 29, 0, 179); // Address of Modbus Slave device

ModbusIP mb; // ModbusIP object

void setup() {
Serial.begin(115200);

WiFi.begin("");

while (WiFi.status() != WL_CONNECTED) {
delay(500);
Serial.print(".");
}

Serial.println("");
Serial.println("WiFi connected");
Serial.println("IP address: ");
Serial.println(WiFi.localIP());

mb.client();
}

bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) { // Modbus Transaction callback
if (event != Modbus::EX_SUCCESS) // If transaction got an error
Serial.printf("Modbus result: %02X\n", event); // Display Modbus error code
if (event == Modbus::EX_TIMEOUT) { // If Transaction timeout took place
mb.disconnect(remote); // Close connection to slave and
mb.dropTransactions(); // Cancel all waiting transactions
}
return true;
}

bool res[COUNT] = {true };

void loop() {
if (!mb.isConnected(remote)) { // Check if connection to Modbus Slave is established
mb.connect(remote); // Try to connect if no connection
Serial.print(".");
}
if (!mb.writeCoil(remote, REG, res, COUNT, cb)) // Try to Write array of COUNT of Coils to Modbus Slave
Serial.print("#");
mb.task(); // Modbus task
delay(50); // Pushing interval
}

emelianov, 2022-08-25:
E4 is Timeout error. E6 caused by reconnection after getting timeout (in `cb` code).
Two possible reasons:
- Communication issues
- Read rate is too high for the server

kiksonix, 2022-08-25:
Thank you. But What can I do now? Where can I slow down rate?

emelianov, 2022-08-28:
Simplest way is to wait response before sending next request. Also you may increase delay time.
```c
void loop() {
  if (!mb.isConnected(remote)) { // Check if connection to Modbus Slave is established 
    mb.connect(remote); // Try to connect if no connection
    Serial.print(".");
  }
  if (uint16_t transaction_id = mb.writeCoil(remote, REG, res, COUNT, cb)) {  // Try to Write array of COUNT of Coils to Modbus Slave
    while (mb.isTransaction(transaction_id) {
      mb.task(); // Modbus task
      yield()
    }
  }
  delay(50); // Pushing interval
}
```

kiksonix, 2022-08-29:
Now I get only E4 error. 

Can you please check the program if I put right data to specificate coils. I want to trigger imput on robot. Robot uses coils adreases 01001...02000 for triggers. I want to write coil 01002. Data type must be bool. 

Thank you so much for help.

Program:`/*
  Modbus-Arduino Example - Modbus IP Client (ESP8266/ESP32)
  Write multiple coils to Slave device

  (c)2019 Alexander Emelianov (a.m.emelianov@gmail.com)
  https://github.com/emelianov/modbus-esp8266
*/

#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP.h>

const int REG = 1002;                    // Modbus Coils Offset
const int COUNT = 1;                    // Count of Coils
IPAddress remote(172, 29, 0, 179);    // Address of Modbus Slave device

ModbusIP mb;  // ModbusIP object

void setup() {
  Serial.begin(115200);
 
  WiFi.begin("...", "...");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  mb.client();
}

bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) { // Modbus Transaction callback
  if (event != Modbus::EX_SUCCESS)                  // If transaction got an error
    Serial.printf("Modbus result: %02X\n", event);  // Display Modbus error code
  if (event == Modbus::EX_TIMEOUT) {    // If Transaction timeout took place
    mb.disconnect(remote);              // Close connection to slave and
    mb.dropTransactions();              // Cancel all waiting transactions
  }
  return true;
}

bool res[COUNT] = {true };

void loop() {
  if (!mb.isConnected(remote)) { // Check if connection to Modbus Slave is established 
    mb.connect(remote); // Try to connect if no connection
    Serial.print(".");
  }
  if (uint16_t transaction_id = mb.writeCoil(remote, REG, res, COUNT, cb)) {  // Try to Write array of COUNT of Coils to Modbus Slave
    while (mb.isTransaction(transaction_id)) {
      mb.task(); // Modbus task
      yield();
    }
  }
  delay(50); // Pushing interval
}`

emelianov, 2022-08-31:
There is no response from the server. Double check network connectivity between client and server. Also probably you need to specify unit_id in writeCoil (you need to refer the server docs).

kiksonix, 2022-08-31:
Is there any chance that you can fix me the program please.

emelianov, 2022-09-01:
Can just suggest
```c
if (uint16_t transaction_id = mb.writeCoil(remote, REG, res, COUNT, cb, 1)) { // where 1 is UnitID
```

closed 2022-09-23 by emelianov