emelianov/modbus-esp8266 #242: Using the librarry with CONTROLLINO MEGA (SN65HVD08 chip)?

skydancerbg, 2022-08-30:
Hello,
Great librarry! I use it all the time with my ESP based projects.

Now i need to make a project with CONTROLLINO MEGA.
Is it possible to use the librarry with CONTROLLINO MEGA? - It has the SN65HVD08 chip bulit in.
Dоes the librarry suport this chip? If yes, how to setup and use?

I understand it is more of a CONTROLLINO question, but in their examples they use some outdated librarry, which has potential problem with buffer overflow. So i better use your librarry if possible.

skydancerbg, 2022-09-01:
I guess, I'll need some help to make it work.

 In CONTROLLINO the SN65HVD08 chip Tx An is connected to Serial3.

The chip has DE and RE (negated) inputs, which are defined in the CONTROLLINO librarry as:
` #define CONTROLLINO_RS485_DE 75
  #define CONTROLLINO_RS485_nRE 76`
The pins are as outputs:
`pinMode(CONTROLLINO_RS485_DE, OUTPUT);
 pinMode(CONTROLLINO_RS485_nRE, OUTPUT);`
To make RS485 serial communication, they use the pins like this:
`voidControllino_RS485TxEnable( void)
{
  /*disable receiver, enable transmitter */
  digitalWrite(CONTROLLINO_RS485_DE, HIGH);
  digitalWrite(CONTROLLINO_RS485_nRE, HIGH);
  return;
}

voidControllino_RS485RxEnable( void)
{
  /*disable trasmitter, enable receiver */
  digitalWrite(CONTROLLINO_RS485_DE, LOW);
  digitalWrite(CONTROLLINO_RS485_nRE, LOW);
  return;
}
`
I tried looking at the Modbus-ESP8266 library code, but I cant figure out how to use the DE and RE for the Modbus communication.
How do I set and use DE, RE in this sample code?

#include <Arduino.h>
#include <Controllino.h>  /* Usage of CONTROLLINO library allows you to use CONTROLLINO_xx aliases in your sketch. */
#include <ModbusRTU.h>
#define SLAVE_ID 1
#define FIRST_REG 0

ModbusRTU mb;
bool coils[20];

bool cbWrite(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  Serial.print("Request result: 0x");
  Serial.println(event, HEX);
  return true;
}
void setup() {
  Serial.begin(115200);
  Serial3.begin(9600, SERIAL_8N1);
  mb.begin(&Serial3);
  mb.setBaudrate(9600);
  mb.master();
}

void loop() {
  if (!mb.slave()) {
    mb.readCoil(1, 1, coils, 20, cbWrite);
  }
  mb.task();
  yield();
  delay(1000);
}`
...
Request result: is  0xE4

emelianov, 2022-09-01:
If it's possible to connect RE and DE together
```c
mb.begin(&Serial3, RE_DE);
```
if not, nconmment in ModbusSettings.h line
```c
#define MODBUSRTU_REDE
```
then independent pins may be used:

```c
mb.begin(&Serial3, DE, RE);
```

skydancerbg, 2022-09-01:
Not possible to connect RE and DE together.
Unconmmented #define MODBUSRTU_REDE
USING
mb.begin(&Serial3, CONTROLLINO_RS485_DE, CONTROLLINO_RS485_nRE);
Still no connection:
Request result: is 0xE4

emelianov, 2022-09-03:
Try to change RE and DE parameters order.

skydancerbg, 2022-09-04:
Still no connection:
Request result: is 0xE4

skydancerbg, 2022-09-04:
Using HTerm and 485 to USB adapter, I can see the librarry sending
01 03 00 01 00 07 55 C8
which is correct when an atempt to read 7 holding registers is made. 
The problem is, it does not recerive the response of ModbusSlave simulator in some reason...

emelianov, 2022-09-07:
Check these initialisation variants:
```c
mb.begin(&Serial3, DE, RE, true);
```
and
```c
mb.begin(&Serial3, DE, RE, false);
```

skydancerbg, 2022-09-08:
It works with: 
mb.begin(&Serial3, DE, RE, true); 

mb.begin(&Serial3, CONTROLLINO_RS485_DE, CONTROLLINO_RS485_nRE, true);

Thank you very much for your help!
You've saved me a lot of time trying to read and understand the details in the library code myself...

closed 2022-09-23 by emelianov