emelianov/modbus-esp8266 #244: Limitation of number read registers

javiergentoo, 2022-09-06:
Hi
First of all, thanks for your contribution.
Do exist any limitation of read register from a slave RTU. I am use the ESP32 as a Master.
I have test a project with more of 40 registers and it didn't read nothing.

Below of 40 registers it is working but above of 40, it is reading all cero.

emelianov, 2022-09-07:
Amount of registers is limited by protocol to 125 if i'mm not mistaken. No additional restrictions.
If Modbus transaction runs with success status I suspect that zeroed registers returned form slave.
Could you provide you ESP32 client-side code to make sure it's Okay ?

javiergentoo, 2022-09-07:
The device have 61 Registers. In this case the ESP re

Of course, 
This is part of my code ( into the setup, I have defined the "mb.master()" 

```
#include <ModbusRTU.h>

ModbusRTU mb;

#define NUM_REG 61
uint16_t registrosTemp[NUM_REG] = {0};

void readAllRegMB()
{
  if (!mb.slave())
  {
    mb.readHreg(245, 0, registrosTemp, NUM_REG, cbReadAll);
    while (mb.slave())
    { 
      mb.task();
      delay(10);
    }
  }
  yield();
}
```

Inside the setup = 
```
  Serial1.begin(9600, SERIAL_8N1, GPIO_RX, GPIO_TX);
  mb.begin(&Serial1);
  mb.master();
```

In this case with 61 Registers, It is return all cero. And if I read device with 20 Registers, it is working fine

emelianov, 2022-09-14:
It should work.
You may enable `MODBUSRTU_DEBUG` in `ModbusSettings.h` to see incoming packets. This will make clear if data from the device is initially filled with zeros.

closed 2022-10-12 by emelianov