emelianov/modbus-esp8266 #247: Write single register issue with function code

simogaspa84, 2022-09-19:
Hello @emelianov  ,

just a question.. 

Why if i run these commands 

  mb.writeHreg(SLAVE_ID, ADDRESS_SINGLE_REGISTER, 0, 1, nullptr, MODBUSIP_UNIT);
  mb.writeHreg(SLAVE_ID,  ADDRESS_SINGLE_REGISTER, 0, 1, nullptr);

The function code for the modbus is 10 

10 1 22 0 1 2 0 0 

instead of  

[image]

emelianov, 2022-09-19:
```c
mb.writeHreg(SLAVE_ID, ADDRESS_SINGLE_REGISTER, 0, nullptr, MODBUSIP_UNIT);
mb.writeHreg(SLAVE_ID, ADDRESS_SINGLE_REGISTER, 0, nullptr);
```

simogaspa84, 2022-09-19:
Just a question @emelianov 

if i run this 

6 1 16 FF FF 

and I got this error from slave
Request result: 0x2

This means that the address of that register is not known ? 

It is correct ?

emelianov, 2022-09-20:
Exactly. Communications are Okay but reg 0x16 doesn't exists on server/slave.