emelianov/modbus-esp8266 #250: ModbusTCP, esp8266 can't connect on device and read registry

tomdelahaba, 2022-10-10:
I have basically the same question as it appeared in here multiple times. I am trying to connect my heat pump and read the registry via ESP8266. I am using the example code to get started, unfortunately the respond is always 0.

I went through the issues in here (mostly the closed ones already) but nothing helped. I made sure that I am setting the slave (unitId) as it was a problem in my python code in the past which is working already. Unfortunately I am not able to get it done with your library. 

My c code (based on your examples with some small changes which can't affect the functionality except the **UnitID** is defined for the readHreg (unfortunately my code never appears in this condition, the device can't login):

```
#include <ESP8266WiFi.h>
#include <ModbusIP_ESP8266.h>

#define PUMP_UNIT_ID 1

const char* ssid = "router_ssid";
const char* password = "pass";

const int REG = 126;               // Modbus Hreg Offset
IPAddress remote(xxx, xxx, xxx, xxx);  // Address of Modbus Slave device
const int LOOP_COUNT = 10;
ModbusIP mb;  //ModbusIP object

void setup() {
  Serial.begin(115200);
 
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  mb.client();
}

uint16_t res = 0;
uint8_t show = LOOP_COUNT;
uint16_t port = 502;

void loop() {
  if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
    Serial.println("Modbus is connected");
    //mb.readHreg(remote, REG, &res);  // Initiate Read Coil from Modbus Slave
    mb.readHreg(remote, REG, &res, 1, nullptr, PUMP_UNIT_ID);
  } else {
    Serial.println("Connecting to modbus");
    mb.connect(remote, port);           // Try to connect if no connection
  }
  mb.task();                      // Common local Modbus task
  delay(100);                     // Pulling interval
  if (!show--) {                   // Display Slave register value one time per second (with default settings)
    Serial.println("Decreasing loops");
    Serial.println(res);
    show = LOOP_COUNT;
  }

  mb.disconnect(remote);
}
``` 

Unfortunately, I aways receive just 0 for the `res`. My python code which is working together with pymodbus:
```
from pymodbus.client import ModbusTcpClient

print('====== STARTING APPLICATION')

port = 502
ip_addr = 'xxx.xxx.xxx.xxx'
slave = 0x01

client = ModbusTcpClient(
    ip_addr,
    port,
    )
client.connect()
result = client.read_input_registers(126, 1, slave=slave) **// THIS SLAVE IS I GUESS THE UNITID ***
print('Value from registry:', result.registers[0])
client.close()
print('====== APP ENDS')
```

emelianov, 2022-10-12:
Basing on your debug output it looks like ESP unable to connect to modbus server device. It's probably because of network issue.

tomdelahaba, 2022-10-12:
> 
To be honest I am sure it is no network issue as I am able to access that device immediately before and immediately after running that code with my python / go code... before and after, as the device (and probably all modbus devices) allows only 1 connection at time. 

Don't you have any other idea, do you? Or is there any possible tracing?

tomdelahaba, 2022-10-13:
@emelianov >
What I found out so far is, it seems that the device is properly connecting as exactly in the "meantime" of processing the commands, from my computer I am not able to access that device and from time to time I get "the target machine is refusing active connection"... While with ESP turned off, I do not get this error and I get proper values.

I tried the same with another computer (connecting 2 computers to the device) and on one of them I get always similar error). So it seems, the ESP is really connected on that device, unfortunately (somehow and for me inconceivably) Instead of reading the registry, the mb.isConnected always return false even when the device seems to be connected (as written above). Well, maybe the device is refusing the connection for the esp as well, can't tell... But this is what I found....

tomdelahaba, 2022-10-13:
Ok found the reason. The very first loop, the esp connects, while I am disconnecting the device at the end of the loop... This is the reason why in new iteration of the loop it has to connect again, which is usually fine but in this case (wonder why) I receive just 0... If I remove the disconnecting, the first iteration of the loop it returns me 0, but in next iterations, the values are properly set...

So I just wonder how I should disconnect the device.

closed 2022-10-14 by tomdelahaba