emelianov/modbus-esp8266 #254: onSetIsts and onSetIreg callback functions

TechLobo, 2022-11-05:
Can't decide if these have been left here for completeness, or if I am missing some use for these - but shouldn't these operations be disallowed?

Not declaring an mb.onSetIsts callback does not appear to interfere with operations - which it shouldn't, as the library should not be processing a 'Write Discrete Input' as modbus does not support that function code.

I note that one of the examples 'onGetShared.ino' provides a write callback function to 'write-protect ' the DI, but this scenario should not occur. If anything the library should raise an 'Illegal Function (01)' exception.

If there is a specific use for these could you please explain as I'm missing it (which may not be surprising)?

TechLobo, 2022-11-06:
Actually having checked the example (onGetShared.ino) properly I note that it is actually using the Coil functionality. So if trying to use this to emulate DI operation then a dummy onSetCoil function would be reasonable - although I'm not sure why the Ists (or DI) functionality wasn't used directly.

Anyway that doesn't change the original question regarding the use of onSetIsts and onSetIreg.

emelianov, 2023-01-01:
For sure the library follows Modbus protocol specification and DI registers are readonly.
From other hand internally you can set value of DI by using `mb.Ireg(ISTS_REG, value)`. That case `onSetIsts` callback make sense and fires.

TechLobo, 2023-01-01:
So are these meant purely for use internally  by the Slave and not associated with a Modbus function code?

emelianov, 2023-01-02:
In case of ISTS (onSetIsts) callback is called on internally initiated events only. In case of, for example, HREG (onSetHreg) callback called on updating Hreg by client requests also.

closed 2023-01-22 by emelianov