emelianov/modbus-esp8266 #259: modbus TCP set slave Id

nex2000, 2022-12-14:
I followed another post where it said to edit this line on the ModbusSettings file line 76.

uint8_t unit=1 // 255

I made this modification but nothing changes. I noticed that the ID can only be changed on push and pull functions. But unfortunately nothing to do even with these commands.

How can I change the slave address number.

Can I get an example ?

Thanks NEX

Merdock1, 2022-12-20:
As far as I know the TCP slave ID is its IP address, unlike the RTU protocol.
Here is more information about it.
[Does Slave ID field have significance in MODBUS TCP](https://control.com/forums/threads/does-slave-id-field-have-significance-in-modbus-tcp.12314/)

emelianov, 2023-01-01:
The library responds on any united when acting as ModbusTCP server.
For the client note last parameter for read\write calls:
```c
uint16_t readCoil(IPAddress ip, uint16_t offset, bool* value, uint16_t numregs = 1, cbTransaction cb = nullptr, uint8_t uint = MODBUSIP_UNIT);
```

closed 2023-01-22 by emelianov