emelianov/modbus-esp8266 #261: Modbus Register 30013

Feierprinz, 2022-12-18:
Hi,

I have an issue with the following Code:
`#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <SoftwareSerial.h>
#include <ModbusRTU.h>
#include "private.h"

SoftwareSerial S(D3, D2);  // we need one serial port for communicating with RS 485 to TTL adapter

#define SLAVE_ID 2

ModbusRTU mb;

WiFiClient espClient;
PubSubClient client(espClient);
long lastMsg = 0;
char msg[50];
int value = 0;

// **************************************************
void setup() {
  Serial.begin(9600);

  setup_wifi();
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);

  S.begin(9600, SWSERIAL_8N1);  // software serial attached to the MAX485 module
  mb.begin(&S, D0);             // RE/DE connected to D0 of ESP8266

  mb.slave(SLAVE_ID);

  // add a series of Input registers
  mb.addIreg(0);
  mb.addIreg(1);
  mb.addIreg(2);
  mb.addIreg(3);
  mb.addIreg(4);
  mb.addIreg(5);
  mb.addIreg(6);
  mb.addIreg(7);
  mb.addIreg(8);
  mb.addIreg(9);
  mb.addIreg(10);
  mb.addIreg(11);
  mb.addIreg(12);
  mb.addIreg(13);

  // set Input register values
  mb.Ireg(0, 250);   // line to neutral volts (integer-part)
  mb.Ireg(1, 400);   // line to neutral volts (fractional-part)
  mb.Ireg(2, 10);    // Current Amps  (integer-part)
  mb.Ireg(3, 54);    // Current Amps (fractional part)
  mb.Ireg(4, 2500);  // Active power Watts (integer part)
  mb.Ireg(5, 76);    // Active power Watts (fractional part)
  mb.Ireg(6, 2639);  // Apparent power VoltAmps (integer part)
  mb.Ireg(7, 216);   // Apparent power VoltAmps (float part)
  mb.Ireg(8, 2375);  // Reactive power VAr (integer part)
  mb.Ireg(9, 5);     // Reactive power VAr (fractional part)
  mb.Ireg(10, 0);    // Power factor (none) (integer part)
  mb.Ireg(11, 90);   // Power factor (none) (fractional part)
  mb.Ireg(12, 30);   // Phase angle (Degree) (integer-part) - ** this is not a
                     // calculated value but my device doesnt actially use it
  mb.Ireg(13, 1);    // Phase angle (Degree) (fractional-part) - ** this is not a
                     // calculated value but my device doesnt actially use it
}

// **************************************************
void setup_wifi() {
  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

// **************************************************
// steps to take when a power value is recieved via MQTT

void callback(char* topic, byte* message, unsigned int length) {
  Serial.print("Message arrived on topic: ");
  Serial.print(topic);
  Serial.print(": ");

  // convert each byte to a char and build a string containing the mqtt message
  String messageTemp;
  for (int i = 0; i < length; i++) {
    // Serial.print((char)message[i]);
    messageTemp += (char)message[i];
  }
  Serial.println(messageTemp);  // print the topic message (it's a string at this point)

  // convert the stirng (messageTemp) to an integer. I'm ignoring the fractional part so 123.456 will return 123
  int power_in_watts;
  const char* string1 = messageTemp.c_str();
  power_in_watts = atoi(string1);
  Serial.printf("Converted to int: %d \n", power_in_watts);

  // update the Active power Watts (integer part)
  mb.Ireg(4, power_in_watts);

  // If a message is received on the topic shellies/shellyem-house/emeter/0/power
  // if (String(topic) == mqtt_topic) {
  //   //Serial.println("We have message!");
  //   // determine is the power value is negative, ie we are exporting power to the grid
  //   if (messageTemp == "on") {
  //     Serial.println("on");
  //   } else if (messageTemp == "off") {
  //     Serial.println("off");
  //   }
  // }
}

// **************************************************
void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect(clientID, mqtt_username, mqtt_password)) {
      Serial.println("Connected to MQTT Broker!");
      // Subscribe
      client.subscribe(mqtt_topic);
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}

// **************************************************
void loop() {
  if (!client.connected()) {
    reconnect();
  }
  client.loop();
  mb.task();
  yield();
}`

In these I must set a value to Register 30013 (Watt-Values) from an MQTT-Request. The last Part works fine and my Inverter receive via your Modbus a "Handshake" that the inverter removes her "Exclamation-Mark" from his Display but he don`t receive any Value for Watts (Register 30013 - Growatt SPH -Inverter)

Can anyone help me? Is the Ireg(4) = Register 30013 ? Questions over questions

Thanks and BR
Swen

PS.: my private.h:
`// Replace the next variables with your SSID/Password combination
const char* ssid = "Fritz!Box";
const char* password = "PASSWORD";

// MQTT setup
const char* mqtt_server = "192.168.0.46";                           // IP of the MQTT broker
const char* mqtt_username = "swen";                                 // MQTT username
const char* mqtt_password = "PASSWORD";                             // MQTT password
const char* clientID = "fakesdm";                                   // MQTT client ID
// const char* mqtt_topic = "shellies/shellyem-house/emeter/0/power";  // MQTT topic to get power reading from
const char* mqtt_topic = "shellies/shellyem3-244CAB435993/emeter/0/power";
`

emelianov, 2023-01-01:
> In these I must set a value to Register 30013 (Watt-Values) from an MQTT-Request. The last Part works fine and my Inverter receive via your Modbus a "Handshake" that the inverter removes her [...]

Depending on register numbering convention used in documentation register 30013 might mean reg(13) or reg(12).

bemilamedia, 2023-01-14:
I have the same problem here. The Growatt solar inverter talks to the original smart meter via ModbusRTU/RS845.

There are different data/PDFs.
The smartmeter sends on the registers 300013,30015,10017 the watts of the individual phases, i.e. power consumption in watts per phase. On register 30053 the balanced total value is sent.
So if you have 
P1 = 100 reference 
P2 = 500 reference
P3 = -700 Generation
this results in the following picture for register 30053:

30053 = 100+500-700 = -100 watts
and for

30013: 100 (watts)
30015: 500 (watts)
30017: -700 (watts)

The question now is, how do I declare these registers 30013/15/17/53 correctly?

Because from the code mentioned by "Feierprinz" registers are always declared via
`mb.InReg(Register number, value)`

Additionally the registers always consist of one LoByte and HighByte.

I have declared it with 

` mb.Ireg(30013, 1000); // Phase 1 Power in Watt (integer part)
  mb.Ireg(30014, 10); // Phase 1 Power in Watt (fractional part)
  mb.Ireg(30015, 1000); // Phase 2 Power in Watt (integer part)
  mb.Ireg(30016, 10); // Phase 1 Power in Watt (fractional part)
  mb.Ireg(30017, 1000); // Phase 3 Power in Watt (integer part)
  mb.Ireg(30018, 10); // Phase 1 Power in Watt(fractional part)
  mb.Ireg(30053, 3000); // Total Power in Watt (integer part)
  mb.Ireg(30013, 30); // total Power in Watt (fractional part)
`
tried. 
But the inverter immediately goes into error 401 = Error with communication between the inverter an "Smartmeter"
When I use the original code, like the code from firepring, there is no error 401.
So I think it must be the correct addressing of the registers.

Here are two more links to the respective log data / registers etc.
Original smartmeter which is recommended:
[https://bg-etech.de/download/manual/SDM630-Modbus-V2.pdf](https://bg-etech.de/download/manual/SDM630-Modbus-V2.pdf)
Growatt PDF:
[https://forum.iobroker.net/assets/uploads/files/1550402978050-growatt-pv-inverter-modbus-rs485-rtu-protocol-v3.05-%E5%AE%A2%E6%88%B7%E7%89%88.pdf](https://forum.iobroker.net/assets/uploads/files/1550402978050-growatt-pv-inverter-modbus-rs485-rtu-protocol-v3.05-%E5%AE%A2%E6%88%B7%E7%89%88.pdf)

emelianov, 2023-01-21:
In case if you are right about way of data coding (I mean spliced int and fractured parts) registers declaration should be
```c
mb.Ireg(12, 1000); // Phase 1 Power in Watt (integer part) 
mb.Ireg(4, 10); // Phase 1 Power in Watt (fractional part) 
mb.Ireg(14, 1000); // Phase 2 Power in Watt (integer part) 
mb.Ireg(15, 10); // Phase 1 Power in Watt (fractional part) 
mb.Ireg(16, 1000); // Phase 3 Power in Watt (integer part) 
mb.Ireg(17, 10); // Phase 1 Power in Watt(fractional part) 
mb.Ireg(52, 3000); // Total Power in Watt (integer part) 
mb.Ireg(53, 30); // total Power in Watt (fractional part)
```

closed 2023-03-31 by emelianov