emelianov/modbus-esp8266 #262: TCP client ESP32 loses information depending on delay

MatheusCaltran, 2023-01-03:
hi,
I am testing TCP/IP communication between two ESP32s, one being the server (send the Hreg Register) the other the client (will receive the Hreg Register) the information it is sending is a random number from 0 - 2000.

when performing this test I verified that depending on the delay in mb.task(); it varies the transmission speed, so I did a test with the 100ms delay and noticed that the reading losses by the client are huge and can reach 6 seconds (repeatedly showing the last number that was received) as can be seen in the video on the left the server(COM7) and on the right the client(COM5).

https://user-images.githubusercontent.com/88856093/210440034-96222561-f3e4-4716-98f2-77644fd679b3.mp4

I performed a test with 250ms/500ms and this loss decreased but it still exists (in the video I only test it with 250ms).

https://user-images.githubusercontent.com/88856093/210440426-8c303707-8a80-464d-b106-3570adab8cb0.mp4

then I performed the tests with 800ms and until it worked well.

https://user-images.githubusercontent.com/88856093/210440821-258440bf-79d4-4d04-b485-b84d825130e3.mp4

So with that I would like to know if there is any way to send information with a delay of around 500ms without this loss happening, I believe that this loss comes from reading the ESP32 client because I have tested the ESP32 server with a simulator CLIENT on pc in 100ms and it worked great.

https://user-images.githubusercontent.com/88856093/210456420-aceadbd5-04e7-47b9-be6d-0b2a6c4092a8.mp4

and also if I leave the client's delay smaller than the server's, the same information loss behavior occurs. I know that 100ms is a very high speed, but I would like at least 500ms works fine.

emelianov, 2023-01-04:
I suspect that you are flooding write requests from sender side. For the first try do not sent next request until get response on previous one. Something like:
```c
uint32_t trans = 0;
void loop() {
  if (!mb.isTransaction(trans)) {
    trans = mb.writeHreg(...);
  }
  mb.task();
}
```

MatheusCaltran, 2023-01-04:
Thanks for the quick response

I made the change on the client:

      uint16_t trans = mb.readHreg(remote, REG, &res);
      Serial.print("Receive:");Serial.println(res);

      while (mb.isTransaction(trans)) { // Check if transaction is active
          mb.task();
  }

now it's working fine. But I have some doubts if I need several Read Register needs to be done (I don't think so):

    uint16_t trans = mb.readHreg(remote, REG, &res);
    uint16_t trans1 = mb.readHreg(remote, REG1, &res1);

    Serial.print("Receive:");Serial.println(res);
     Serial.print("Receive1:");Serial.println(res1);

     while (mb.isTransaction(trans)) {
     while (mb.isTransaction(trans1)) {
         mb.task();
    }
    }

Or I just need to do isTransaciton with only one remote information instead of all,
and if I have to get this information from another remote(IP- another server) what needs to be done

thank you very much

emelianov, 2023-01-05:
Indeed `.readHreg`/`.writeHreg`/etc returns transaction id (or `0` if unable to initiate request). With ModbusTCP it's okay to multiple requests just keep in mind that `.readHreg`/`.writeHreg`/etc just sends the request. The response is processed by `.task()`. Count of simultaneous requests is limited by 16 by default. As for you code I suggest
```c
uint16_t trans = mb.readHreg(remote, REG, &res);
uint16_t trans1 = mb.readHreg(remote, REG1, &res1);
// <- at this point requests just dent to remote and no response data in res and res1 yet
 while (mb.isTransaction(trans) || mb.isTransaction(trans1)) { // Wait transactions to complete
     mb.task();
}
// <- Only at this point you have res and res1 filled by returned data
Serial.print("Receive:");Serial.println(res);
Serial.print("Receive1:");Serial.println(res1);

```

MatheusCaltran, 2023-01-05:
OK thank you

MatheusCaltran, 2023-05-24:
It looks like I'm suffering the same thing here, only now I'm using a TCP simens s71200 client(Write) and ESP32 server(Read)

I am doing TPC communication between a PLC and an ESP32, I would like to know if there is any way to improve communication because sometimes the ESP keeps showing the repeated value, it varies from 2 seconds to 10 seconds, then it returns to normal

Does it have something like Transactional.ino for the server? I'm using:
variable = mb.Hreg(Address of the register to read)

if I request the IP of the CLP in readHreg(.....) I can't communicate with it just using mb.Hreg