emelianov/modbus-esp8266 #264: Examples to read uint32_t 32 bits modbus register

legaxus, 2023-01-05:
Hi,
Appreciate if you could show examples on reading uint32_t 32 bits modbus register?

emelianov, 2023-01-05:
[Article](https://github.com/emelianov/modbus-esp8266/issues/158)

legaxus, 2023-01-05:
uint16_t res[REG_COUNT];
readHreg(SLAVE_ID, FIRST_REG, res, REG_COUNT);
from the modbus library
so it will be something like this
firstvalue = res[0]
second = res[1]
then combine?

emelianov, 2023-01-06:
```c
union f_2uint
    {
        float f;
        uint16_t i[2];
    };
f_2uint raw_reg;
readHreg(SLAVE_ID, FIRST_REG, (uint16_t*)raw_reg, 2);
```
And probably play with byte order then.

legaxus, 2023-01-08:
union f_2uint
    {
        float f;  <-
        uint16_t i[2];
    };
f_2uint raw_reg;
readHreg(SLAVE_ID, FIRST_REG, (uint16_t*)raw_reg, 2);

if its register uint32_t but not float, so I need to change "int f;" ? am I right?

emelianov, 2023-01-08:
Sure. `uint32_t f;`
It was just copy-paste mistake.

legaxus, 2023-01-11:
ok, thanks, give a try,

closed 2023-03-31 by emelianov

fuzonmedia, 2024-01-04:
> ```c
> union f_2uint
>     {
>         float f;
>         uint16_t i[2];
>     };
> f_2uint raw_reg;
> readHreg(SLAVE_ID, FIRST_REG, (uint16_t*)raw_reg, 2);
> ```
> 
> And probably play with byte order then.

I am trying to use this using TCP. 

`   // start test
    union f_2uint
    {
        float f;
        uint16_t i[2];
    };
    
f_2uint raw_reg;
mb.readHreg(remote, 156, (uint16_t*)raw_reg, 2);`

It gives this below mentioned error. Can you help ?

`invalid cast from type 'loop()::f_2uint' to type 'uint16_t* {aka short unsigned int*}'`