emelianov/modbus-esp8266 #266: direction pin

hedva92, 2023-01-20:
Hi, could you implement direction control pin to begin? I have a max485 board installed, but I don't have hardware control of the data direction, I need to add the direction control pin attribute to the begin function, which is mapped to the physical port of the microprocessor.

emelianov, 2023-01-21:
This is already have implemented:

```c
bool begin(SoftwareSerial* port, int16_t txPin=-1, bool direct=true);
bool begin(HardwareSerial* port, int16_t txPin=-1, bool direct=true);
bool begin(Stream* port, int16_t txPin=-1, bool direct=true);
```

Assing Serial port. txPin controls transmit enable for MAX-485. Pass direct=false if txPin uses inverse logic.

rob040, 2023-03-13:
For "txPin", a better name would be "txEnablePin" to describe its function. The TX pin is normally the UART transmit data pin, not an enable pin.
For "direct", a better name would be "txEnableInvert" to describe its function.

emelianov, 2023-04-01:
Naming changed in 0bc538d1eb43b2416733154c2fbc546c15b68858. Just `txEnableDirect` used as using Invert notation will break API.

closed 2023-04-01 by emelianov

rob040, 2023-12-15:
Agree about "txEnableInvert" would break the API, however, "Direct" is a non-descript name. 
The logic alternative would be "txEnableNoninvert"