emelianov/modbus-esp8266 #274: Чтение массива регистров.

kurnatskyi, 2023-03-13:
Добрый вечер. Возможно ли как то считать массив регистров?

Допустим есть такой массив с адресами регистров.
`uint16_t addr[] = {2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 70, 71, 72, 73, 74, 75, 76, 77, 90, 91, 92, 93, 94, 95, 100, 101, 102, 103, 104, 105, 106, 120, 121, 122, 123, 124, 125, 126}`

Есть массив с кол-вом регистров:
`uint16_t qty[] = {17, 8, 6, 7, 7};`

 mb.readHreg(SLAVE_ID, addr[0], &rhr[0], qty[0], cb); 
    while(mb.slave()) {                                                                 
      mb.task();                                                                        
      delay(30);                                                               
    }          

Читает первые 17 регистров.
Потом необходимо заново вызивать функцию readHreg и task для чтения следующих?

rob040, 2023-03-17:
Translation?

kurnatskyi, 2023-03-18:
Ok.

kurnatskyi, 2023-03-18:
Ok.
```c
void loop() {
  if (!mb.slave()) {
    mb.readHreg(SLAVE_ID, 2, &rhr[0], 17, cb); 
    while(mb.slave()) {                                                                
      mb.task();                                                                                                                                                   
      delay(30);                                                                      
    }                                                                                  
    mb.readHreg(SLAVE_ID, 70, &rhr[17], 8, cb);                                     
    while(mb.slave()) {                                                               
      mb.task();                                                                                                                                                  
      delay(30);                                                                       
    } 
  }
}
  
``` 
can i read all registers in one call task()?

emelianov, 2023-03-31:
No. You can't place non-sequental registers within single modbus request.
--
Нет. Modbus не может включать непоследовательные регистры в рамках одного запроса.

closed 2023-04-02 by kurnatskyi