emelianov/modbus-esp8266 #289: Device Connected, but "MODBUS Exception Response from Slave Device" at Modscan

arimukhlas, 2023-05-02:
Hi @emelianov  thanks for great library

I'm trying based on [sample Library ](https://github.com/emelianov/modbus-esp8266/blob/ea2bd0e506b2c32755ca86edcd2ea95acc1b0855/examples/TCP-ESP/IP-server-AnalogInput/IP-server-AnalogInput.ino) 
```c++
#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else //ESP32
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>

//Modbus Registers Offsets
const int SENSOR_HREG = 100;

//ModbusIP object
ModbusIP mb;

long ts;

void setup() {
    Serial.begin(115200);

    WiFi.begin("MySSID", "MyPass");
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");  
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());

    mb.server();		//Start Modbus IP
    // Add SENSOR_HREG register - Use addHreg()
    mb.addHreg(SENSOR_HREG);

    ts = millis();
}

void loop() {
   //Call once inside loop() - all magic here
   mb.task();

   //Read each two seconds
   if (millis() > ts + 2000) {
       ts = millis();
       //Setting raw value (0-1024)
       mb.Hreg(SENSOR_HREG, 5);
   }
   delay(10);
}
``` 

after upload sketch, esp got IP 192.168.43.224

[image]

Then I try to see the data value using Modscan Software

[image]

but I dont see the data value

[image]

Is my procedure correct, or something I miss?
How to see data value on Modscan? Thanks

Merdock1, 2023-06-21:
With this code, I get results without problems
[https://github.com/Merdock1/DataConversion/blob/main/src/Examples/ExampleAll.ino](https://github.com/Merdock1/DataConversion/blob/main/src/Examples/ExampleAll.ino)

[image]

Merdock1, 2023-06-21:
I think your problem is in the number of records, your code only has one holding record at position 100 and you ask the software to read 10 records from position 100, I think that is the reason for your error, you must tell it to read 1 register at position 100

closed 2023-10-14 by emelianov