emelianov/modbus-esp8266 #293: pullHreg - how to modify local registers

chrgraf, 2023-06-06:
Dear Alexander,

first thanks for this cool lib.

From reading the doc, I am having issues to understand the pullHreg method: 
uint16_t pullHreg(IPAddress ip, uint16_t from, uint16_t to, uint16_t numregs = 1,..

1) meaning of from/to: do those values denote the starting and the ending-range when reading registers? e.g .would mb_real_meter.pullHreg(real_meter_ID, 243, 299);  read all Holding  registers from starting-range 243 upto 299 (so in total 46 regs)?
2) the docs state: Result is saved to local registers. Method returns corresponding transaction id. [ip/from] or [ip/offset] - slave, [to] or [startreg] - local
Kindly asking how to access and modify those registers please

thanks

christian

duluthmachineworks, 2023-08-29:
Regarding number two - I got this to work in my code. Basically, you have to establish a local register on the master device with the addHreg() function. Then you pull the result from the remote register, to your local register. So `pullHreg(slave_address, slave_hreg, local_hreg)`. Then you can read the local hreg via the .Hreg() function: `modbusObject.Hreg(local_hreg)`.

Someone else feel free to correct me if I'm doing this wrong.

closed 2023-10-14 by emelianov

chrgraf, 2023-11-04:
thanks alot!